jimy-byerley/pymadcad #2: Problems with using booleans(Intersection)

hit-the-luke, 2021-03-04:
Hi,

I just installed pymadcad and going my first steps and I cant figure out how the boolean functions are supposed to work. The closest I have come to my expected result was with the following code:

m1 = madcad.brick(width=madcad.vec3(2))
m2 = m1.transform(madcad.vec3(1, -0.5, 1))
print(m1)
print(m2)
m3 = madcad.boolean.intersectwith(m1, m2)
print(madcad.Mesh(m3))

Am I using it wrong?

the corresponding output was the following:

Mesh(
  points= [dvec3(            1,           -1,           -1 ),
                      ...
           dvec3(           -1,            1,           -1 )],
  faces=  [(0, 1, 2),
           ...,
           (4, 0, 3),
           (4, 3, 7)],
  tracks= [0,
           0,
           1,
             ...
           5,
           5],
  groups= [None,
           None,
           None,
           None,
           None,
           None],
  options= {})
Mesh(
  points= [dvec3(            2,         -1.5,            0 ),
           dvec3(            2,         -1.5,            2 ),
           dvec3(            0,         -1.5,            2 ),
           dvec3(            0,         -1.5,            0 ),
           dvec3(            2,          0.5,            0 ),
           dvec3(            2,          0.5,            2 ),
           dvec3(            0,          0.5,            2 ),
           dvec3(            0,          0.5,            0 )],
  faces=  [(0, 1, 2),
           (0, 2, 3),
           ...
           (4, 3, 7)],
  tracks= [0,
           0,
           1,
           ...
           5,
           5],
  groups= [None,
           None,
           None,
           None,
           None,
           None],
  options= {})
Mesh(
  points= [((8, 9), 9),
           ((10, 9), 10),
           ((15, 13), 2),
           ((10, 13), 10),
           ((15, 17), 2),
           ((8, 17), 8)],
  faces=  [],
  tracks= [],
  groups= [],
  options= {})

Thanks a lot for in advance and for doing this project in general. It sounds really promising

Lukas

hit-the-luke, 2021-03-04:
Thank you very much for the quick Answer!!
I could reproduce your solution. I am interested where you found the Algorithm you are referring to (syandana). I tried to have a look for it on the web, but google left me hanging there...

jimy-byerley, 2021-03-04:
You're welcome !

> I tried to have a look for it on the web, but google left me hanging there...

Oh that's ... because I created that algorithm  ;)
It's slightly different that usual O(n**2) algorithms because it's using spatial hashing and few other techniques.

I'm planning to put de detail in the 'algorithms' section of the documentation, but I didn't had time yet.  I'm mostly busy with with a software I develop on top of pymadcad

jimy-byerley, 2021-03-09:
I worked hard the two last days to fix all the known bugs of the boolean operations:
- crossing edges at intersections
- intersection when the intersection edge was already existing
- some other weird ones

so everything should work fine now with the repo version
Those fixes will be in the next release :)

closed 2021-03-15 by jimy-byerley