jimy-byerley/pymadcad #3: Function "blendpair" in "blending.py" issues

bourbonut, 2021-03-31:
Hello, I was trying some functions to make a part.

madcad version 0.7

Here is my code:

```python
from madcad import *

circle1 = Circle((vec3(0,0,0),vec3(0,0,1)),5)
circle2 = Circle((vec3(0,0,0),vec3(0,0,1)),20)
z = blendpair(circle1, circle2, tangents = "straight")
z = extrusion(vec3(0,0,10), z)
show([z])
```
The following image is the result:
[image]

The hole is not empty and there are some strange lines near the central circle as you can see.
Maybe is my code wrong ?

jimy-byerley, 2021-03-31:
Hello dear user.
Your code is not wrong at all.  This is the correct usage except `blendpair` doesn't handle closed lines yet. It's working fine for two curves, but not for loops.

So the matching between the points of each lines are made quite randomly (like one point of a circle matched to the opposite point of the other circle)

Well I just worked to add the ability to handle closed lines:   (commit 1d8af58bdcb6f7df991d5ef0113b7d1942b484d3)

This will be in the next release ;)  you can also use it immediately if you clone the latest github version

jimy-byerley, 2021-03-31:
Oh and by the way, thanks for having reported this :+1:

closed 2021-03-31 by jimy-byerley

jimy-byerley, 2021-04-01:
I didn't noticed yesterday, but there is an other issue.
Not a bug this time, but a small little thing in your code:  you are blending between 2 circles with the same winding (both normals in the same direction, `vec3(0,0,1)` in this case)

Everywhere in madcad, faces outlines are oriented using their [winding](https://en.wikipedia.org/wiki/Winding_number). It means in this case that the inner circle must rotate in the opposite direction to the outer circle. 
Else the inner circle is considered to be an outer line as the other one, leading to an incorrect output.

This should work better:
```python
from madcad import *

circle1 = Circle((vec3(0,0,0),vec3(0,0,-1)),5)   # inner rotates negatively around z
circle2 = Circle((vec3(0,0,0),vec3(0,0,1)),20)    # outer rotates positively around z
z = blendpair(circle1, circle2, tangents = "straight")
z = extrusion(vec3(0,0,10), z)
show([z])
```
[image]