jimy-byerley/pymadcad #6: Error while using pip install pymadcad

FaezAlkadi, 2021-06-18:
Hello 
Thanks for the great work. 
I tried to install `pymadcad` using `pip install pymadcad ` but unfortunately I faced the following error:

` Running setup.py install for pymadcad ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\faez\pycharmprojects\pyshop\venv\scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Use
rs\\Faez\\AppData\\Local\\Temp\\pip-install-ul46jryj\\pymadcad_caf2118f1998466b8c8789fadbc9af8c\\setup.py'"'"'; __file__='"'"'C:\\Users\\Faez\\AppData\\
Local\\Temp\\pip-install-ul46jryj\\pymadcad_caf2118f1998466b8c8789fadbc9af8c\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.
path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();
exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Faez\AppData\Local\Temp\pip-record-fkz6pgb4\install-record.txt' --single-versi
on-externally-managed --compile --install-headers 'c:\users\faez\pycharmprojects\pyshop\venv\include\site\python3.8\pymadcad'
         cwd: C:\Users\Faez\AppData\Local\Temp\pip-install-ul46jryj\pymadcad_caf2118f1998466b8c8789fadbc9af8c\
    Complete output (77 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.8
    creating build\lib.win-amd64-3.8\madcad
    copying madcad\asso.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\blending.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\boolean.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\common.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\constraints.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\cut.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\displays.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\gear.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\generation.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\hashing.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\io.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\joints.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\kinematic.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\mathutils.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\mesh.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\nprint.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\primitives.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\rendering.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\reverse.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\scheme.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\selection.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\settings.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\text.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\triangulation.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\__init__.py -> build\lib.win-amd64-3.8\madcad
    creating build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\annotation.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\axis.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\font.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\ghost.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\glowenvelope.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\ident.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\pointhalo.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\scheme-ghost.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\scheme-ident.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\scheme-uniform.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\simple-solid.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\solid-old.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\solid.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\uniformcolor.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\viewgrid.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\wire.frag -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\annotation.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\axis-ident.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\axis.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\font.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\glowenvelope.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\object-ident.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\object-item-ident.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\pointhalo-ident.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\pointhalo.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\scheme.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\solid.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\uniformcolor.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\viewgrid.vert -> build\lib.win-amd64-3.8\madcad\shaders
    copying madcad\shaders\wire.vert -> build\lib.win-amd64-3.8\madcad\shaders
    creating build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\point.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-blue-dense.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-blue.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-test.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-violet-dense.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-violet-saturated.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-violet.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-white.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\core.pyx -> build\lib.win-amd64-3.8\madcad
    copying madcad\core.c -> build\lib.win-amd64-3.8\madcad
    running build_ext
    building 'madcad.core' extension
    error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-buil
d-tools/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\faez\pycharmprojects\pyshop\venv\scripts\python.exe' -u -c 'import io, os, sys, setuptools, tok
enize; sys.argv[0] = '"'"'C:\\Users\\Faez\\AppData\\Local\\Temp\\pip-install-ul46jryj\\pymadcad_caf2118f1998466b8c8789fadbc9af8c\\setup.py'"'"'; __file_
_='"'"'C:\\Users\\Faez\\AppData\\Local\\Temp\\pip-install-ul46jryj\\pymadcad_caf2118f1998466b8c8789fadbc9af8c\\setup.py'"'"';f = getattr(tokenize, '"'"'
open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r
\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Faez\AppData\Local\Temp\pip-record-fkz6pgb4\i
nstall-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\faez\pycharmprojects\pyshop\venv\include\site\python3.8\pym
adcad' Check the logs for full command output.
`

How can I resolve this issue. 
Thank you so much

jimy-byerley, 2021-06-18:
Hello @FaezAlkadi !

This installation error is because madcad contains some internal modules written in C code that must be compiled. And it appears There is not the Visual C++ compiler installed on your machine.

A module compilation for python is a very simple operation on Linux, but is  I worked one week ago to build precompiled packages for the next version of madcad ! So in few days you should be able to install it without

FaezAlkadi, 2021-06-19:
Hello @jimy-byerley 
Thank you for your detailed reply. I would appreciate it if you could let me knew when you get the package ready. 

Best of luck.

jimy-byerley, 2021-06-19:
Hello again, here comes [v0.8](https://github.com/jimy-byerley/pymadcad/releases/tag/v0.8)
I made a compiled package for windows 64bit, already available on pypi, So you should be able to install it smoothly now:
```shell
pip install pymadcad>=0.8
```

I'm still working on adding more recompiled packages but fighting against cloud compilation systems ;)

Let me know if it works for you !

FaezAlkadi, 2021-06-20:
Hello @jimy-byerley 

Unfortunately its still not working. 
I tried installing it but got the following error again

```
osoft Windows [Version 10.0.19041.1052]
(c) Microsoft Corporation. All rights reserved.

(venv) C:\Users\Faez\Desktop\Python>pip install pymadcad
Collecting pymadcad
  Using cached pymadcad-0.8.tar.gz (1.5 MB)
Requirement already satisfied: pyglm>=1.2 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (2.2.0)
Requirement already satisfied: moderngl>=5.6 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (5.6.4)
Requirement already satisfied: numpy>=1.1 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (1.20.3)
Requirement already satisfied: scipy>=1.3 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (1.6.3)
Requirement already satisfied: PyQt5>=5 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (5.15.4)
Requirement already satisfied: Pillow>=5.4 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (8.2.0)
Requirement already satisfied: cython>=0.29 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (0.29.23)
Requirement already satisfied: pyyaml>=5 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from pymadcad) (5.4.1)
Requirement already satisfied: glcontext<3,>=2 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from moderngl>=5.6->pymadcad) (2.3.4)
Requirement already satisfied: PyQt5-sip<13,>=12.8 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from PyQt5>=5->pymadcad) (12.9.0)
Requirement already satisfied: PyQt5-Qt5>=5.15 in c:\users\faez\pycharmprojects\pyshop\venv\lib\site-packages (from PyQt5>=5->pymadcad) (5.15.2)
Using legacy 'setup.py install' for pymadcad, since package 'wheel' is not installed.
Installing collected packages: pymadcad
    Running setup.py install for pymadcad ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\users\faez\pycharmprojects\pyshop\venv\scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Use
rs\\Faez\\AppData\\Local\\Temp\\pip-install-jhlnh1oc\\pymadcad_0563471ff4f349f78234b2e78d35bfd7\\setup.py'"'"'; __file__='"'"'C:\\Users\\Faez\\AppData\\
Local\\Temp\\pip-install-jhlnh1oc\\pymadcad_0563471ff4f349f78234b2e78d35bfd7\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.
path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();
exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Faez\AppData\Local\Temp\pip-record-_h70kuli\install-record.txt' --single-versi
on-externally-managed --compile --install-headers 'c:\users\faez\pycharmprojects\pyshop\venv\include\site\python3.8\pymadcad'
         cwd: C:\Users\Faez\AppData\Local\Temp\pip-install-jhlnh1oc\pymadcad_0563471ff4f349f78234b2e78d35bfd7\
    Complete output (93 lines):
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.8
    creating build\lib.win-amd64-3.8\madcad
    copying madcad\annotations.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\asso.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\auto.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\blending.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\boolean.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\common.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\constraints.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\container.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\cut.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\derived.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\displays.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\editors.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\gear.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\generation.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\hashing.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\io.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\joints.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\kinematic.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\mathutils.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\mesh.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\nprint.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\post.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\primitives.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\rendering.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\reverse.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\scheme.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\selection.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\settings.py -> build\lib.win-amd64-3.8\madcad
    copying madcad\soft.py -> build\lib.win-amd64-3.8\madcad
[... 43 lines cut ...]
    copying madcad\textures\skybox-blue-dense.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-blue.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-test.png -> build\lib.win-amd64-3.8\madcad\textures
    copying madcad\textures\skybox-violet-dense.png -> build\lib.win-amd64-3.8\madcad\textures
[... cut ...]
open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r
\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Faez\AppData\Local\Temp\pip-record-_h70kuli\i
nstall-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\faez\pycharmprojects\pyshop\venv\include\site\python3.8\pym
adcad' Check the logs for full command output.
```

jimy-byerley, 2021-06-20:
It still tries to compile the source instead of using the precompiled package. What windows do you have ? Is not you processor 64bits ?

FaezAlkadi, 2021-06-20:
I am using windows 10 with 67-bit operating system, x64-based processor.

jimy-byerley, 2021-06-20:
Oh and I should have asked before, which version of python ? I only compiled for 3.9 as it is the current stable

jimy-byerley, 2021-06-20:
Very well, I finally got [travis](https://travis-ci.com/github/jimy-byerley/pymadcad) to work and build properly ! 
So as long as you have python>=3.7 it should be easy to install by now :slightly_smiling_face:

FaezAlkadi, 2021-06-20:
Hi, 
Yes 
it does seem working now 
I was wondering
Do you have a function for reading STL files and returning `[points, ...], [triangles, ...]`

Thank you

jimy-byerley, 2021-06-20:
Good to know that it works !

> Do you have a function for reading STL files

Different topic but yes, the submodule [madcad.io](https://pymadcad.readthedocs.io/en/latest/reference/io.html) does that sort of thing:
```python
>>> import madcad as mc
>>> mc.read('my/file.stl')
<Mesh ...>
```
It internally relies on module `numpy-stl`

closed 2021-06-20 by jimy-byerley

FaezAlkadi, 2021-06-20:
Why can't we say `import pymadcad as mc` instead of `import madcad as mc`

because I tried to read stl and got this following error:

```
 Traceback (most recent call last):
  File "C:\Users\Faez\PycharmProjects\pythonProject\offset_Substrate.py", line 4, in <module>
    import madcad as mc
  File "C:\Users\Faez\PycharmProjects\pythonProject\venv\lib\site-packages\madcad\__init__.py", line 135, in <module>
    from . import (
  File "C:\Users\Faez\PycharmProjects\pythonProject\venv\lib\site-packages\madcad\mesh.py", line 33, in <module>
    from . import displays
  File "C:\Users\Faez\PycharmProjects\pythonProject\venv\lib\site-packages\madcad\displays.py", line 8, in <module>
    from .rendering import Display, overrides, writeproperty
  File "C:\Users\Faez\PycharmProjects\pythonProject\venv\lib\site-packages\madcad\rendering.py", line 51, in <module>
    from . import settings
  File "C:\Users\Faez\PycharmProjects\pythonProject\venv\lib\site-packages\madcad\settings.py", line 66, in <module>
    config = os.getenv('HOME')+'/.config/madcad/pymadcad.yaml'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

Sorry for the question 
But I am actually a beginner with python.

jimy-byerley, 2021-06-21:
It seems I forgot some "commits" in my release ...  So I made a new one, if you can try once again ?

> But I am actually a beginner with python.

Don't worry for this, I was also a beginner in the beginning ;)

FaezAlkadi, 2021-06-21:
Thanks for the reply 
Unfortunately it still has the same error :)

FaezAlkadi, 2021-06-22:
Great !
Now it is working.
However, There is a small bug regarding file type:
When I use higher case letter (STL) 
```
import madcad as mc
part = mc.read("0001-STL files/1x2x3 mm.STL")
```
I get the following error:

```
  File "C:\Users\Faez\anaconda3\lib\site-packages\madcad\io.py", line 29, in read
    raise FileFormatError('no read function available for format '+type)
madcad.io.FileFormatError: no read function available for format STL
``` 

But when I use lower case  letter (stl) :
```
import madcad as mc
part = mc.read("0001-STL files/1x2x3 mm.stl")
```

It runs perfect 👍

jimy-byerley, 2021-06-22:
Good to know for that upper case issue, I will fix that in a next version :+1: