jimy-byerley/pymadcad #7: Import module Error

XBlayz, 2021-11-06:
I have installed the module on my pc (Windows 10 x64 - Python 3.10) and I tried to run the Sample that I found on the main page of the GitHub repository, but when I tray to run it I get this error:

**"[WinError 193] %1 is not a valid win32 application"**

What could it be?

jimy-byerley, 2021-11-06:
Hello and welcome there !

This quite an odd error since that kind of error is usually not platform or version specific.
- Could you please put the complete backtrace of that error ?
- How did you installed the module ? (pip ?)

XBlayz, 2021-11-06:
[image]
_This is the error VS Code displays to me._

- Yes, I have installed it using `pip install pymadcad`

jimy-byerley, 2021-11-06:
VS Code should be truncating the backtrace in some way so we do not get the full message.
This should be reporting at what place in madcad the error is encountered :thinking: 
Can you try to `import madcad` from the python commandline instead ?

I never tested madcad on python 3.10 yet
And I'm not on windows nor python 3.10 at the moment, so I cannot attempt to reproduce right now

XBlayz, 2021-11-06:
This is the result.
```
>>> import madcad
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "E:\Programmi HDD\Python\lib\site-packages\madcad\__init__.py", line 135, in <module>
    from . import (
  File "E:\Programmi HDD\Python\lib\site-packages\madcad\mesh.py", line 33, in <module>
    from . import displays
  File "E:\Programmi HDD\Python\lib\site-packages\madcad\displays.py", line 8, in <module>
    from .rendering import Display, overrides, writeproperty
  File "E:\Programmi HDD\Python\lib\site-packages\madcad\rendering.py", line 51, in <module>
    from . import settings
  File "E:\Programmi HDD\Python\lib\site-packages\madcad\settings.py", line 66, in <module>
    config = os.getenv('HOME')+'/.config/madcad/pymadcad.yaml'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

XBlayz, 2021-11-06:
I think I have solved the problem.
I simply replaced the path of `pymadcad.yaml` by deleting `os.getenv ('HOME')`, as if the file is not found it is automatically generated, and now it seems to work.
In case you can tell me more let me know, thanks a lot for the help anyway.

closed 2021-11-06 by XBlayz

jimy-byerley, 2021-11-07:
Ok I know what happened: It seems `pip` installed an old version that is `pymadcad-0.8.3`
It should have installed it because the latest version `pymadcad-0.9.0` has no precompiled package for windows with `python-3.10` (and you might have no visual studio toolchain installed to build from source)

The issue with `os.getenv('HOME')` was solved in `pymadcad-0.9.0`  [by this commit](https://github.com/jimy-byerley/pymadcad/commit/e3845aba3b92f7fbe9a0ed6cdba616b90960913e)
It simply replace `getenv('HOME')` by a windows equivalent, so the config file is put at the right place in config folders.

I will try to generate prebuilt packages for windows, so you will benefit of all improvements of `pymadcad-0.9.0`

XBlayz, 2021-11-07:
Ok, thanks for the clarification.