jimy-byerley/pymadcad #9: Parasolid save of the CAD

Boreas3, 2021-11-19:
Hello everyone,

I am searching for the possibility to save the created CAD into a Parasolid file?

Thanks for your help.

jimy-byerley, 2021-11-19:
Hello and welcome here !
By Parasolid files I assume you mean `.step` files or solid edge files or solidworks files ?  If yes, I have bad news.

I only have a little experience with these file formats (I only know the solidworks side I mean). But I suspect those formats will never be compatible with madcad :-/ ... And there is strong reasons:
The trick is that 
- most parametric cad softwares (including Parasolid derivatives like solidworks or solidedge, Catia, or OpenCascade) rely on [boundary representations](https://en.wikipedia.org/wiki/Boundary_representation) to model parts. It means every edge or face is of a known familly with a known equation. 
- madcad on the other hand relies on [triangular meshes](https://en.wikipedia.org/wiki/Triangle_mesh) to model parts. Here, no face or edge has specific equation or anything, it's just ... points coordinates.

So sadly, madcad **cannot** output something else than meshes, and most parasolid derivatives **cannot** load them. 

There is advantages and downsides to both, but mainly here is a summary:
triangular meshes are
- less precise than B-rep
- more flexible and general than B-rep

Just out of curiosity: what exactly was your purpose in exporting to a parasolid file ?

Boreas3, 2021-11-19:
Hello!
I thank you for your response.
In fact, I would have wanted to use this python library to generate CAD that could be in the CFD solver (CFX from Ansys).
This would have been an alternative to the parametric cad software that I currently am using for my work.
For some geometries (like a volute for turbocompressor/turboexpander), the way cad is generated with madcad was attracting me. 
But If you are telling me that it would be very tricky to generate such file formats, then I will stick to the conventional way to make CAD!

Again, I thank you of having taken the time to answer me.

Have a great day!

PeterMoresco, 2021-11-19:
If you wish, there's a way to convert from mesh(STL, OBJ) to STEP using FreeCAD, it's very straightforward, and also FreeCAD is free and OpenSource. One can generate the geometries there too, or using OpenSCAD one can generate the solid right away. If you're using the Ansys suite, SpaceClaim is able to convert meshes into solids.

jimy-byerley, 2021-11-20:
Greate news @PeterMoresco ! I wasn't aware of this function of FreeCAD and of the existence of SpaceClaim. It is good news that there is holes in that wall between B-rep and mesh

Boreas3, 2021-11-20:
I thank you both for your responses.
indeed, it seems that there are ways to perform what I am looking for!

jimy-byerley, 2021-11-20:
I close this topic, but feel free to reopen if you see new problems in the conversion.
Also tell me if it works fine :)

closed 2021-11-20 by jimy-byerley