jimy-byerley/pymadcad #16: failed to write ply

donaldlee2008, 2022-01-08:
Traceback (most recent call last):
  File "test5.py", line 17, in <module>
    write(original, 'test_io.ply')
  File "D:\Program Files (x86)\Microsoft Visual Studio\Shared\Anaconda3_64\envs\
pyqtdemo\lib\site-packages\pymadcad-0.10.2-py3.7-win-amd64.egg\madcad\io.py", li
ne 38, in write
    raise FileFormatError('no write function available for format '+type)
madcad.io.FileFormatError: no write function available for format ply

here is the code 

from madcad.mathutils import vec3
from madcad.mesh import Web
from madcad.generation import extrusion
from madcad.io import *
from madcad.generation import *

original = extrusion(vec3(0,0,0.5), Web(
        [vec3(1,1,0), vec3(-1,1,0), vec3(-1,-1,0), vec3(1,-1,0)],
        [(0,1), (1,2), (2,3), (3,0)],
        [0,1,2,3],
        ))
print(filetype('data/test_io.ply' ))
# test ply
type=filetype('data/test_io.ply' )
writer = globals().get(type+'_write')
print("writer:",writer)
write(original, 'test_io.ply')
ply = read('data/test_io.ply')
ply.check()
assert ply.issurface()

# test stl
write(original, 'data/test_io.stl')
stl = read('data/test_io.stl')
stl.check()
assert stl.issurface()

jimy-byerley, 2022-01-08:
Hello there

`no write function available for format ply` means the module `plyfile` used to read/write files of this kind is not found. As it is an optional dependency, you must have forgot to install it. The same applies for stl files.
You can install it with either:
```shell
pip install pymadcad[ply,stl]
```
or
```shell
pip install plyfile numpy-stl
```

donaldlee2008, 2022-01-09:
it works .thank you

closed 2022-01-09 by donaldlee2008