jimy-byerley/pymadcad #17: thicken does not work

donaldlee2008, 2022-01-11:
from madcad import *
from madcad.io import *
from madcad.generation import *
print("runing here ")
surface=read("data/aa.stl")
print("runing here 0")
surface = thicken(surface,0.1 ,0.5) 
print("runing here 1")
write(surface, 'data/A12.stl')
print("runing here 2")
show([surface])

>python thickentest.py
  runing here
  runing here 0

that mean I can't run thicken(surface,0.1 ,0.5) .
the  STL file is fine. I can open it on other software

donaldlee2008, 2022-01-11:
inflateoffsets, inflate have the same problem

donaldlee2008, 2022-01-11:
there is no any error or warning. just no result

jimy-byerley, 2022-01-11:
Can you please format the code you post ? It's hard to read without a monospaced font,with no tabs nor colors.
Can you also post a `.stl` file having this issue ? so I could try to reproduce the issue.  `inflate` always worked in my use cases.

donaldlee2008, 2022-01-11:
please see file attached. code and stl file
[code_and_data.zip](https://github.com/jimy-byerley/pymadcad/files/7849925/code_and_data.zip)

jimy-byerley, 2022-01-12:
Hello, the probleme is pretty simple in fact:  

- first thing first, your model is of size `1000`, so inflating/thickening of `0.1` doesn't make a big difference ;)

    ```python
    show([ surface, note_bounds(surface) ])
    ```
 
    [image]

-  `inflate` works by moving points along their normal. But a `.stl` file only gives separated triangles (no information about which  triangle is connected to which). Thus the points are moved separately and the triangles get floating apart from each others.

    ```python 
    surface = read('data/aa.stl')
    inflated = inflate(surface, 5)

    show([surface], options={'display_groups': False})
    ```
    
    [image]

   To make the triangles connected, you can just merge the points that are close enough

   ```python 
    surface = read('data/aa.stl')
    surface.mergeclose()
    inflated = inflate(surface, 5)

    show([surface], options={'display_groups': False})
    ```

    ![image](https://user-images.githubusercontent.com/7940753/149219478-e681fb55-e984-458e-90e1-c04dbfec6810.png)

Note that the edges visible on the mesh in the first picture of the image are surface contours. Because by default, edges between adjacent triangles are not displayed (for convenience). So you can easily check when one imported mesh has badly connected components.

closed 2022-01-14 by jimy-byerley