jimy-byerley/pymadcad #24: Boolean Operation cut_mesh

mia0120, 2022-02-23:
I need a rectangular flat surface with a hole. 
First I create a rectangular surface with a flatsurface command and a disk with Circle command.
Then using cut_mesh, I try to empty first rectangular surface. But cut_mesh command turns the part to a tuple object.
How can I create this flat surface?
[image]

jimy-byerley, 2022-02-23:
That's because [`cut_mesh`](https://pymadcad.readthedocs.io/en/latest/reference/boolean.html#madcad.boolean.cut_mesh) cuts the mesh faces, but does remove the faces inside the holes (you can see the result below)

```python
result, frontier = cut_mesh(rectangle, cylinder)
```
[image]

## Solution with a boolean operation
[`pierce`](https://pymadcad.readthedocs.io/en/latest/reference/boolean.html#madcad.boolean.pierce) is the boolean operation that does what you want
```python
rectangle = flatsurface(wire([
	vec3(-1,-1,0), 
	vec3(1,-1,0), 
	vec3(1,1,0),
	vec3(-1,1,0),
	]))
cylinder = extrusion(Z, Circle((O,Z), 0.5))   # note that the cylinder face normals are oriented outside (the surface color is bright) so the faces to remove are inside the cylinder

result = pierce(rectangle, cylinder)
show([result], options={'display_wire':True})
```
[image]

## Solution with contours

However, the boolean operations are quite expensive in term of performance for any mesh, and to use is for simple planar operations like this can be avoided in this case.
You can directly use [`flatsurface`](https://pymadcad.readthedocs.io/en/latest/reference/generation.html#madcad.generation.flatsurface) with the appropriate `Web`

```python
rectangle = wire([
		vec3(-1,-1,0), 
		vec3(1,-1,0), 
		vec3(1,1,0),
		vec3(-1,1,0),
		]) .close()
circle = Circle((O,-Z), 0.5)   # note that the circle axis is flipped (Z is negative) to ensure that the contour winding is consistent. This is the equivalent of the face normals in the previous solution

result = flatsurface(web([
	rectangle,
	circle,
	]))
show([result], options={'display_wire':True})
```

This will give the exact same result

Hope this help !

jimy-byerley, 2022-02-23:
Nonetheless I agree that the boolean operation docs are lacking nice screenshots to explain this ;)

closed 2022-02-25 by jimy-byerley