jimy-byerley/pymadcad #33: Errors on MacOSX

jeromegout, 2022-03-20:
Hi, 

Is there a known issue using pymadcad on MacOsX?
I'm trying to run an example from your "examples" folder and I receive the following errors:

```
qt.qpa.openglcontext: Could not create NSOpenGLContext with shared context, falling back to unshared context.
Traceback (most recent call last):
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 1099, in paintGL
    self.render()
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 981, in render
    ViewCommon.render(self)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 712, in render
    self.uniforms['proj'] = proj = self.projection.matrix(w/h, self.navigation.distance)
ZeroDivisionError: division by zero
[1]    8975 abort      /Users/jeromegout/dev/david/implantology/.venv/bin/python 
```

It seems that is related to OpenGL.
Any clue or advice

Thanks in advance
Jerome

jimy-byerley, 2022-03-20:
Hi @jeromegout 

No known issue as far as I know, that's the first one (though I don't know about other mac users). the `ZeroDivisionError` seems to happen because of the `w/h` ratio in case of `h = 0`, can you check that ?
What happens exactly ? you are trying to call `madcad.show` and you get an empty window and this error ?

jeromegout, 2022-03-21:
Hi Jimy,

I see the error trace, I don't know how I could check the value of h within the lib? (sorry I'm a python noob)
I've tried the same example (nut.py) on Windows it works fine.
I've only noticed (on both platforms) the wrong value 'CG' for the last parameter of solve function, at line 46.
```
Traceback (most recent call last):
  File "C:\perso\dev\python\madcad\nut.py", line 46, in <module>
    solve(csts, fixed=[O,X,Y,Z,P0,P5], precision=1e-6, method='CG')
  File "C:\perso\dev\python\madcad\lib\site-packages\madcad\constraints.py", line 195, in solve
    return Problem(constraints, fixed).solve(*args, **kwargs)
  File "C:\perso\dev\python\madcad\lib\site-packages\madcad\constraints.py", line 312, in solve
    res = least_squares(evaluate, self.state(),
  File "C:\perso\dev\python\madcad\lib\site-packages\scipy\optimize\_lsq\least_squares.py", line 762, in least_squares
    raise ValueError("`method` must be 'trf', 'dogbox' or 'lm'.")
ValueError: `method` must be 'trf', 'dogbox' or 'lm'.
```
Once I've changed it to 'trf' it works on Windows and put the other error on Mac.
I guess that this example has not been updated after a change in the solve function.

On Mac, nothing happens except the error, no window (even empty) is open, I just get the error.
Thanks

jimy-byerley, 2022-03-25:
> I guess that this example has not been updated after a change in the solve function.

That's true ! the `'CG'` value is something that changed in the last release, I forgot to update the examples ... 

For the crash, the problem is I do not have an Apple machine for testing, would you mind if we fix the issue on your side ?

The first thing to do is to locate the installation path of madcad: your log shows it is `/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/`

Things I would like you to test, if you don't mind:

1. print the value of `w,h` in `madcad.rendering.ViewCommon.render()`  (in the file `rendering.py`)

   The modified function should look like this:
   ```python
   def render(self):
    # prepare the view uniforms
    w, h = self.fb_screen.size
    print('size', w, h)
    self.uniforms['view'] = view = self.navigation.matrix()
    self.uniforms['proj'] = proj = self.projection.matrix(w/h, self.navigation.distance)
    self.uniforms['projview'] = proj * view
    self.fresh.clear()
    
    # call the render stack
    self.scene.render(self)
   ```
   
2. try to ignore rendering if `h = 0`
 
   ```python
   def render(self):
    # prepare the view uniforms
    w, h = self.fb_screen.size
    print('size', w, h)
    if h*w:
      self.uniforms['view'] = view = self.navigation.matrix()
      self.uniforms['proj'] = proj = self.projection.matrix(w/h, self.navigation.distance)
      self.uniforms['projview'] = proj * view
      self.fresh.clear()
      
      # call the render stack
      self.scene.render(self)
   ```
 
3. try disabling `Qt.AA_ShareOpenGLContexts`

   This should be done in `madcad.rendering.show()`  (in the file `rendering.py`) by commenting the following line:
   ```python
   #QApplication.setAttribute(Qt.AA_ShareOpenGLContexts, True)
   ```

jeromegout, 2022-03-26:
Hi @jimy-byerley 

Here is the result of these three changes:

```python
(.venv) ➜  implantology git:(master) ✗ /Users/jeromegout/dev/david/implantology/.venv/bin/python /Users/jeromegout/dev/david/implantology/src/nut.py
/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/scipy/optimize/_lsq/least_squares.py:110: UserWarning: Setting `gtol` below the machine epsilon (2.22e-16) effectively disables the corresponding termination condition.
  warn("Setting `{}` below the machine epsilon ({:.2e}) effectively "
qt.qpa.openglcontext: Could not create NSOpenGLContext with shared context, falling back to unshared context.
size 0 0
qt.qpa.openglcontext: Could not create NSOpenGLContext with shared context, falling back to unshared context.
size 0 0
`

It's worth noting that there is now an empty window opening.
Hope this helps
Thanks
Jerome

jimy-byerley, 2022-03-26:
Hello Jerome
It's very odd that Qt is still complaining on the OpenGL context sharing now that you disabled the `AA_ShareOpenGLContexts` flag ... I have to investigate more ...

jimy-byerley, 2022-03-26:
Can you tell me the output of the command
```shell
$ glxinfo
```
Also what version of OSX are you running on ?

More things to test:

4. disable the openGL core profile requirement in `madcad.rendering.View.__init__`

   It is known to sometimes make problems on windows and OSX
   Comment the following line
   ```python
   #fmt.setProfile(QSurfaceFormat.CoreProfile)
   ```
   This is to be tests with *3.*
   
5. higher the requestion openGL version to 4.1

    it seems OSX only runs openGL 2.1 and 4.1
    To try it you need to change the requirement at the top of `rendering.py`
    ```python
    #opengl_version = (3,3)
    opengl_version = (4,1)
    ```
    This is to be tests with and without *4.*, but with *3.*

jeromegout, 2022-03-26:
here is the glxinfo output:

```
name of display: /private/tmp/com.apple.launchd.PDg1SF5OTf/org.macosforge.xquartz:0
display: /private/tmp/com.apple.launchd.PDg1SF5OTf/org.macosforge.xquartz:0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_multisample, GLX_EXT_import_context, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_OML_swap_method, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_import_context, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_MESA_multithread_makecurrent, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig
OpenGL vendor string: ATI Technologies Inc.
OpenGL renderer string: AMD Radeon R9 M370X OpenGL Engine
OpenGL version string: 2.1 ATI-4.7.103
OpenGL shading language version string: 1.20
OpenGL extensions:
    GL_APPLE_aux_depth_stencil, GL_APPLE_client_storage, 
    GL_APPLE_element_array, GL_APPLE_fence, GL_APPLE_float_pixels, 
    GL_APPLE_flush_buffer_range, GL_APPLE_flush_render, 
    GL_APPLE_object_purgeable, GL_APPLE_packed_pixels, GL_APPLE_pixel_buffer, 
    GL_APPLE_rgb_422, GL_APPLE_row_bytes, GL_APPLE_specular_vector, 
    GL_APPLE_texture_range, GL_APPLE_transform_hint, 
    GL_APPLE_vertex_array_object, GL_APPLE_vertex_array_range, 
    GL_APPLE_vertex_point_size, GL_APPLE_vertex_program_evaluators, 
    GL_APPLE_ycbcr_422, GL_ARB_color_buffer_float, GL_ARB_depth_buffer_float, 
    GL_ARB_depth_clamp, GL_ARB_depth_texture, GL_ARB_draw_buffers, 
    GL_ARB_draw_elements_base_vertex, GL_ARB_draw_instanced, 
    GL_ARB_fragment_program, GL_ARB_fragment_program_shadow, 
    GL_ARB_fragment_shader, GL_ARB_framebuffer_object, 
    GL_ARB_framebuffer_sRGB, GL_ARB_half_float_pixel, 
    GL_ARB_half_float_vertex, GL_ARB_imaging, GL_ARB_instanced_arrays, 
    GL_ARB_multisample, GL_ARB_multitexture, GL_ARB_occlusion_query, 
    GL_ARB_pixel_buffer_object, GL_ARB_point_parameters, GL_ARB_point_sprite, 
    GL_ARB_provoking_vertex, GL_ARB_seamless_cube_map, GL_ARB_shader_objects, 
    GL_ARB_shader_texture_lod, GL_ARB_shading_language_100, GL_ARB_shadow, 
    GL_ARB_shadow_ambient, GL_ARB_sync, GL_ARB_texture_border_clamp, 
    GL_ARB_texture_compression, GL_ARB_texture_compression_rgtc, 
    GL_ARB_texture_cube_map, GL_ARB_texture_env_add, 
    GL_ARB_texture_env_combine, GL_ARB_texture_env_crossbar, 
    GL_ARB_texture_env_dot3, GL_ARB_texture_float, 
    GL_ARB_texture_mirrored_repeat, GL_ARB_texture_non_power_of_two, 
    GL_ARB_texture_rectangle, GL_ARB_texture_rg, GL_ARB_transpose_matrix, 
    GL_ARB_vertex_array_bgra, GL_ARB_vertex_blend, 
    GL_ARB_vertex_buffer_object, GL_ARB_vertex_program, GL_ARB_vertex_shader, 
    GL_ARB_window_pos, GL_ATI_blend_equation_separate, 
    GL_ATI_blend_weighted_minmax, GL_ATI_separate_stencil, 
    GL_ATI_texture_compression_3dc, GL_ATI_texture_env_combine3, 
    GL_ATI_texture_float, GL_ATI_texture_mirror_once, GL_EXT_abgr, 
[... 409 lines cut ...]
0x129  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2  0  0  0  0  0  0  0 0 None
0x12a  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2  0  0  0  0  0  0  8 1 None
0x12b  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 16  0  0  0  0  0  0 0 None
0x12c  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 16  0  0  0  0  0  8 1 None
0x12d  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 24  0  0  0  0  0  0 0 None
0x12e  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 24  0  0  0  0  0  8 1 None
0x12f 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2  0  0  0  0  0  0  0 0 None
0x130 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2  0  0  0  0  0  0  8 1 None
0x131 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 16  0  0  0  0  0  0 0 None
0x132 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 16  0  0  0  0  0  8 1 None
0x133 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 24  0  0  0  0  0  0 0 None
0x134 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 24  0  0  0  0  0  8 1 None
0x135  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2  0  8  0  0  0  0  0 0 None
0x136  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2  0  8  0  0  0  0  8 1 None
0x137  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 16  8  0  0  0  0  0 0 None
0x138  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 16  8  0  0  0  0  8 1 None
0x139  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 24  8  0  0  0  0  0 0 None
0x13a  0 tc  1  16  0 r  y .   5  5  5  1 .  .  2 24  8  0  0  0  0  8 1 None
0x13b 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2  0  8  0  0  0  0  0 0 None
0x13c 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2  0  8  0  0  0  0  8 1 None
0x13d 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 16  8  0  0  0  0  0 0 None
0x13e 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 16  8  0  0  0  0  8 1 None
0x13f 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 24  8  0  0  0  0  0 0 None
0x140 24 tc  1  32  0 r  y .   8  8  8  8 .  .  2 24  8  0  0  0  0  8 1 None
```

jeromegout, 2022-03-26:
My version of OSX is 12.2.1

jeromegout, 2022-03-26:
test 4 says:

```
ValueError: Requested OpenGL version 330, got version 0
```

Is it possible I don't have OpenGL on my machine?

jimy-byerley, 2022-03-26:
That's bad news :-/
Your glxinfo says that you do have openGL but only in versions 1.4 and 1.2. And madcad needs >=3.3 ...

Yeah FreeCAD is built with Qt as madcad, but using VTK for its 3D graphics (and VTK relies on very old openGL version as far as I know) where madcad directly rely on openGL with its modern features.  I designed madcad with those modern features to program, maintain more easily and have nicer renderings. Compatibility shouldn't have been an issue since 3.3 is now supported by all the GPUs since 2010-2015.

I'm afraid It won't work on your current OSX installation... For what I have read, OSX now supports openGL 4.1 and so you OSX 12 should be fine with madcad, but that support is limited to the GPUs that can run it. 
Is your machine GPU too old to support it ?

jimy-byerley, 2022-03-26:
Here is for instance my own result of `glxinfo`
You will notice that there is lines saying `GLX version string: 4.6`, ensuring madcad to work

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_ATI_pixel_format_float, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_multithread_makecurrent, 
    GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_NV_float_buffer, 
    GLX_OML_swap_method, GLX_OML_sync_control, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, GLX_MESA_swap_control, 
    GLX_OML_swap_method, GLX_OML_sync_control, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon Vega 8 Graphics (raven, LLVM 13.0.0, DRM 3.41, 5.13.0-37-generic) (0x15d8)
    Version: 22.0.0
    Accelerated: yes
    Video memory: 2048MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 1620 MB, largest block: 1620 MB
    VBO free aux. memory - total: 2981 MB, largest block: 2981 MB
    Texture free memory - total: 1620 MB, largest block: 1620 MB
[... 304 lines cut ...]
    GL_NV_conditional_render, GL_NV_draw_buffers, GL_NV_fbo_color_attachments, 
    GL_NV_image_formats, GL_NV_pixel_buffer_object, GL_NV_read_buffer, 
    GL_NV_read_depth, GL_NV_read_depth_stencil, GL_NV_read_stencil, 
    GL_OES_EGL_image, GL_OES_EGL_image_external, 
    GL_OES_EGL_image_external_essl3, GL_OES_EGL_sync, 
    GL_OES_compressed_ETC1_RGB8_texture, GL_OES_copy_image, GL_OES_depth24, 
    GL_OES_depth_texture, GL_OES_depth_texture_cube_map, 
    GL_OES_draw_buffers_indexed, GL_OES_draw_elements_base_vertex, 
    GL_OES_element_index_uint, GL_OES_fbo_render_mipmap, 
    GL_OES_geometry_point_size, GL_OES_geometry_shader, 
    GL_OES_get_program_binary, GL_OES_gpu_shader5, GL_OES_mapbuffer, 
    GL_OES_packed_depth_stencil, GL_OES_primitive_bounding_box, 
    GL_OES_required_internalformat, GL_OES_rgb8_rgba8, GL_OES_sample_shading, 
    GL_OES_sample_variables, GL_OES_shader_image_atomic, 
    GL_OES_shader_io_blocks, GL_OES_shader_multisample_interpolation, 
    GL_OES_standard_derivatives, GL_OES_stencil8, GL_OES_surfaceless_context, 
    GL_OES_tessellation_point_size, GL_OES_tessellation_shader, 
    GL_OES_texture_3D, GL_OES_texture_border_clamp, GL_OES_texture_buffer, 
    GL_OES_texture_cube_map_array, GL_OES_texture_float, 
    GL_OES_texture_float_linear, GL_OES_texture_half_float, 
    GL_OES_texture_half_float_linear, GL_OES_texture_npot, 
    GL_OES_texture_stencil8, GL_OES_texture_storage_multisample_2d_array, 
    GL_OES_texture_view, GL_OES_vertex_array_object, GL_OES_vertex_half_float, 
    GL_OES_viewport_array
```

jeromegout, 2022-03-26:
My GPU is:

AMD Radeon R9 M370X 2 Go
Intel Iris Pro 1536 Mo

My Machine is a Macbook pro 2015 ... old machine yes

jeromegout, 2022-03-26:
I have to admit that I don't really understand all details of OpenGL.
I see this stackoverflow answer:

https://stackoverflow.com/a/26982694

But I don't understand if I can update or not.

jimy-byerley, 2022-03-27:
For what I read, you cannot update anything since you are  on a macbook. However they are saying that `glxinfo` is not using NSOpenGL to retreive its informations, but Qt uses NSOpenGL according to the error messages you get.
So maybe you should take a look at what [opengl-viewer](https://apps.apple.com/us/app/opengl-extensions-viewer/id444052073?mt=12) they recommend to see what NSOpenGL supports ?

jeromegout, 2022-03-27:
Hi @jimy-byerley 

I did install and launch GLViewer. 
First when I open it, I see that the OpenGL version is 2.1 and without touching nothing after a couple of seconds it changes to 4.1.
I can test openGL and it seems that 4.1 is installed.
I have 3 renderers:
 - Intel Iris pro (graphical shipset of my CPU I guess)
 - AMD Radeon R9 M370X
 - Apple Software
I can select the renderer I want to see details.
All of them are marked as 4.1

Here is the report for AMD Radeon renderer:

```
Renderer: AMD Radeon R9 M370X 
OpenGL Engine Vendor: ATI Technologies Inc. 
Version: 4.1 ATI-4.8.13 
Device: MacBookPro11,5 
Shading language version: 4.10

Core features

 v3.0 (100 % - 23/23) 
v3.1 (100 % - 7/7) 
v3.2 (100 % - 10/10) 
v3.3 (100 % - 10/10) 
v4.0 (100 % - 14/14) 
v4.1 (100 % - 7/7) 
v4.2 (0 % - 0/13)
 v4.3 (0 % - 0/20)
 v4.4 (0 % - 0/10) 
v4.5 (0 % - 0/11) 
v4.6 (0 % - 0/11) 
vARB 2015 (0 % - 0/12)

OpenGL driver version check (Current: 4.1 ATI-4.8.13, Latest known: ): 
Latest version of display drivers found According the database, you are running the latest display drivers for your video card. 
```

I've never used openGL myself so I'm lost in all this details.
Anyway, thank you for your time

jimy-byerley, 2022-03-27:
Well then I suppose Qt should be working with it then ... Strange that test 3 did not work ...
Do you think you cat retry it with:

6. explicitely disable context sharing

   replace the following line by the second
   ```python
   #QApplication.setAttribute(Qt.AA_ShareOpenGLContexts, True)
   QApplication.setAttribute(Qt.AA_ShareOpenGLContexts, False)
   ```
   
   This should be the default with Qt by have no idea of what else we can try ...

jeromegout, 2022-03-27:
With False doesn't work, but with True (QApplication.setAttribute(Qt.AA_ShareOpenGLContexts, True))
I have an other error:

```
/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/scipy/optimize/_lsq/least_squares.py:110: UserWarning: Setting `gtol` below the machine epsilon (2.22e-16) effectively disables the corresponding termination condition.
  warn("Setting `{}` below the machine epsilon ({:.2e}) effectively "
qt.qpa.openglcontext: Could not create NSOpenGLContext with shared context, falling back to unshared context.
size 0 0
size 800 800
tried to display <madcad.rendering.Displayable object at 0x11783bdf0>
Traceback (most recent call last):
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 688, in update
    self.displays[key] = scene.display(obj)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 603, in display
    disp = obj.display(self)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 643, in display
    return self.build(scene, *self.args, **self.kwargs)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/scheme.py", line 437, in __init__
    self.disp = build(1), build(-1)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/scheme.py", line 436, in build
    return scene.display(sch)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 601, in display
    disp = obj.display(self, obj)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/scheme.py", line 233, in __init__
    self.components = [(space,scene.display(obj))       for space,obj in sch.components]
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/scheme.py", line 233, in <listcomp>
    self.components = [(space,scene.display(obj))       for space,obj in sch.components]
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 603, in display
    disp = obj.display(self)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/text.py", line 48, in display
    return TextDisplay(scene, self.position, self.text, self.size, self.color, self.align)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/text.py", line 77, in __init__
    self.fonttex, self.fontalign = scene.ressource(('fonttex', size), load)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 584, in ressource
    res = func(self)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/text.py", line 75, in load
    img, align = create_font_texture(ImageFont.truetype(ressourcedir+'/NotoMono-Regular.ttf', 2*size))
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/PIL/ImageFont.py", line 855, in truetype
    return freetype(font)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/PIL/ImageFont.py", line 852, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/PIL/ImageFont.py", line 211, in __init__
    self.font = core.getfont(
OSError: cannot open resource
tried to display <madcad.scheme.Scheme object at 0x11783bc10>
Traceback (most recent call last):
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 688, in update
    self.displays[key] = scene.display(obj)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 601, in display
    disp = obj.display(self, obj)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/scheme.py", line 233, in __init__
    self.components = [(space,scene.display(obj))       for space,obj in sch.components]
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/scheme.py", line 233, in <listcomp>
    self.components = [(space,scene.display(obj))       for space,obj in sch.components]
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 603, in display
    disp = obj.display(self)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/text.py", line 48, in display
    return TextDisplay(scene, self.position, self.text, self.size, self.color, self.align)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/text.py", line 77, in __init__
    self.fonttex, self.fontalign = scene.ressource(('fonttex', size), load)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/rendering.py", line 584, in ressource
    res = func(self)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/madcad/text.py", line 75, in load
    img, align = create_font_texture(ImageFont.truetype(ressourcedir+'/NotoMono-Regular.ttf', 2*size))
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/PIL/ImageFont.py", line 855, in truetype
    return freetype(font)
[... cut ...]
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/PIL/ImageFont.py", line 852, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/Users/jeromegout/dev/david/implantology/.venv/lib/python3.9/site-packages/PIL/ImageFont.py", line 211, in __init__
    self.font = core.getfont(
OSError: cannot open resource
```

Size is 800x800, promising.
I did try also to force the openGL version to 4.1 (on top of render.py) without any success.

jeromegout, 2022-03-27:
If you are interested to live debug this situation, we can setup a session where I can share my screen.

jimy-byerley, 2022-03-28:
Interesting to see that enabling a non-supported option makes it work :thinking:  ...

At least I know the cause of this new error: you are missing [`NotoMono-Regular.ttf`](https://github.com/jimy-byerley/pymadcad/blob/master/madcad/NotoMono-Regular.ttf)
So you can fix it by downloading the file and placing it in the madcad folder

I noticed last week it was missing from the latest source code package on pypi. It does affect all text annotations. 

Not sure when I will have time for a live debug.  I will tell you

jeromegout, 2022-03-28:
Ok, right now, no more error of missing font but window remains empty.

jimy-byerley, 2022-03-28:
ah ... is it empty with a pure black background, or is there any color ?

jeromegout, 2022-03-28:
Pure black

jimy-byerley, 2022-03-29:
*"Now there is two of them ..."* :sweat_smile: 
Welcome @granludo I'm sorry about this bug

@jeromegout Some more things to try if you can:

7. redetect the framebuffer when `h = 0`

    Can you insert this instead of `madcad.rendering.View.render`
    ```python
    def render(self):
      self.makeCurrent()
      if self.fb_screen.size == (0,0):
           self.init()
      assert self.fb_screen.size != (0,0)
      ViewCommon.render(self)
    ```

8. can you give me the reports the eventual errors happenning in openGL ?

    Insert the following at the beginning and end of `madcad.rendering.View.render`
    ```python
    def render(self):
      print(self.scene.ctx.error)
      ...
      print(self.scene.ctx.error)
    ```
    and at the end of `madcad.rendering.View.init`
    ```python
    def init(self):
      ...    
      print(self.scene.ctx.error)
    ```

GenieTim, 2023-06-09:
Sorry to pitch in here, (yes, now there's three of them) – was any of the fixes ever fruitfull?

jimy-byerley, 2023-06-09:
Unfortunately, no ... as far as I know. And I still have no time to work on this problem.
If someone reading this thread succeeded in some way, please let us know

Maybe a fallback method for rendering on macos could be offscreen rendering, then copying the GPU buffer to some Qt picture and refresh it in the window ... less efficient of course but this could bypass the issue of Qt with openGL :thinking:

jimy-byerley, 2023-06-17:
Can some mac-osx users here try the [branch](https://github.com/GenieTim/pymadcad/tree/master) of @GenieTim on the same machine they previously had problems to run pymadcad on ?

jimy-byerley, 2024-01-29:
I will assume the fix worked

closed 2024-01-29 by jimy-byerley