jimy-byerley/pymadcad #49: Create a new mesh

Eva-xx, 2022-09-23:
Your work is quite useful and convenient for my research. But I have a trouble now, because of some unknown problem I can only install v0.10.2 at most. If I have a set of n*3 numpy arrays, how can I easily convert them to the format required by faces so that I can construct a mesh by myself?

jimy-byerley, 2022-09-24:
Thanks you :)

# installing pymadcad v0.13

I would strongly recommend you to fix that installation problem, as bugfixes and new features will come on the top of newer versions. the next version after 0.10.2 was 0.11 that brough [arrex](https://github.com/jimy-byerley/arrex) in madcad and changed the way points and faces are represented in memory. Since then, it is much easier and efficient to convert from-to numpy arrays.

In pymadcad>=0.11 you can use `typedlist_to_numpy` and `numpy_to_typedlist` to make conversions:
```python
# from numpy
# depending on the numpy implementation and the CPU type, this could be no-copy
points = np.zeros((n,3), np.float64)
faces = np.zeros((n,3), np.uint32)

m = Mesh(
    points=numpy_to_typedlist(points, vec3),
    faces=numpy_to_typedlist(faces, uvec3),
    tracks=numpy_to_typedlist(tracks, 'I'),    # you may not need this as you may not have surface groups infos from your source ?
    groups=?  # groups is expected to be a python list, put what you want in
    )

# to numpy
# depending on the numpy implementation and the CPU type, this could be no-copy
m = Mesh(...)

points = typedlist_to_numpy(m.points, np.float64)
faces = typedlist_to_numpy(m.faces, np.uint32)
tracks = typedlist_to_numpy(m.tracks, np.uint32)  # you way not need to extract tracks since most 3D computing libraries do not have this notion
# no point to convert groups to numpy since most groups are `None` or `dict` instances
```

How did you tried to install upper versions of pymadcad ?
Can you copy-paste here the result of `pip install pymadcad>=0.13` please ?

# converting mesh from-to numpy arrays in pymadcad v0.10

It depends on the version of [pyglm](https://github.com/Zuzu-Typ/PyGLM/) you have. With old versions of pyglm there was few issues with the conversion from numpy to vectors.

Assuming you have pyglm>=2.5.7 (the latest), you can use the function `glm.array` as intermediate step. But this leads to a lot of copies:
```python
# from numpy
points = np.zeros((n,3), np.float64)
faces = np.zeros((n,3), np.uint32)

m = Mesh(
    points=[vec3(p)    for p in points],
    faces=[tuple(f)     for f in faces],
    tracks=list(tracks),   # you may not need this as you may not have surface groups infos from your source ?
    groups=?  # groups is expected to be a python list, put what you want in

# to numpy
m = Mesh(...)

points = np.asarray(glm.array(m.points))
faces = np.asarray(m.faces)
tracks = np.asarray(m.faces)  # you way not need to extract tracks since most 3D computing libraries do not have this notion
# no point to convert groups to numpy since most groups are `None` or `dict` instances
```

Eva-xx, 2022-09-25:
Thank you for your reply!
I solved the problem just fine using the glm.array function you mentioned!

Regarding the installation of pymadcad v0.13, the result of pip install pymadcad==0.13.0 is as follows：

```
Collecting pymadcad==0.13.0
  Using cached pymadcad-0.13.0.tar.gz (1.4 MB)
Requirement already satisfied: pyglm>=2.5.5 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (2.5.7)
Requirement already satisfied: moderngl>=5.6 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (5.6.4)
Requirement already satisfied: numpy>=1.1 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (1.21.2)
Requirement already satisfied: scipy>=1.3 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (1.7.1)
Requirement already satisfied: PyQt5>=5 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (5.15.7)
Requirement already satisfied: Pillow>=5.4 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (8.4.0)
Requirement already satisfied: pyyaml>=5 in d:\anaconda\lib\site-packages (from pymadcad==0.13.0) (6.0)
Collecting arrex>=0.3.1
  Using cached arrex-0.5.0.tar.gz (197 kB)
    ERROR: Command errored out with exit status 1:
     command: 'D:\Anaconda\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\USER\\App
Data\\Local\\Temp\\pip-install-ru5dt2hh\\arrex_19730870d46c4499930072f02ae7d100\\setup.py'"'"'; __file__='"'"'C:\\Us
ers\\USER\\AppData\\Local\\Temp\\pip-install-ru5dt2hh\\arrex_19730870d46c4499930072f02ae7d100\\setup.py'"'"';f=getat
tr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(comp
ile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\USER\AppData\Local\Temp\pip-pip-egg-info-kpv43cx
f'
         cwd: C:\Users\USER\AppData\Local\Temp\pip-install-ru5dt2hh\arrex_19730870d46c4499930072f02ae7d100\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\USER\AppData\Local\Temp\pip-install-ru5dt2hh\arrex_19730870d46c4499930072f02ae7d100\setup.py",
line 40, in <module>
        long_description = open('README.md').read(),
    UnicodeDecodeError: 'gbk' codec can't decode byte 0x8b in position 3863: illegal multibyte sequence
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/ca/67/2c129af4b4a397a9f758d99e9d01c71014e7670cb6b0073b0d
11832a9e88/arrex-0.5.0.tar.gz#sha256=10ee3405267fb85f781ad1e8658e34ef6ff5bdf0f4d020acb7629e10960ecabc (from https://
pypi.org/simple/arrex/) (requires-python:>=3.8). Command errored out with exit status 1: python setup.py egg_info Ch
eck the logs for full command output.
  Using cached arrex-0.4.1.tar.gz (196 kB)
    ERROR: Command errored out with exit status 1:
     command: 'D:\Anaconda\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\USER\\App
Data\\Local\\Temp\\pip-install-ru5dt2hh\\arrex_7c5cb2422c5e454daa1f0cecab5df97c\\setup.py'"'"'; __file__='"'"'C:\\Us
ers\\USER\\AppData\\Local\\Temp\\pip-install-ru5dt2hh\\arrex_7c5cb2422c5e454daa1f0cecab5df97c\\setup.py'"'"';f=getat
tr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(comp
ile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\USER\AppData\Local\Temp\pip-pip-egg-info-afstc0h
w'
         cwd: C:\Users\USER\AppData\Local\Temp\pip-install-ru5dt2hh\arrex_7c5cb2422c5e454daa1f0cecab5df97c\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\USER\AppData\Local\Temp\pip-install-ru5dt2hh\arrex_7c5cb2422c5e454daa1f0cecab5df97c\setup.py",
line 40, in <module>
        long_description = open('README.md').read(),
    UnicodeDecodeError: 'gbk' codec can't decode byte 0x8b in position 3863: illegal multibyte sequence
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/c8/a0/84ebac8b48c35c167fda9522bc23e142931cc38571e7489d08
dd7a0af1bf/arrex-0.4.1.tar.gz#sha256=1c0cffca90f1e93d0a8dcbfdf448107dc2e7e0cea96d0bbc42e50300273213d3 (from https://
pypi.org/simple/arrex/) (requires-python:>=3.8). Command errored out with exit status 1: python setup.py egg_info Ch
eck the logs for full command output.
  Using cached arrex-0.4.0.tar.gz (196 kB)
    ERROR: Command errored out with exit status 1:
     command: 'D:\Anaconda\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\USER\\App
Data\\Local\\Temp\\pip-install-ru5dt2hh\\arrex_198cb0a2b1904a24a0a5f2bce1f516f6\\setup.py'"'"'; __file__='"'"'C:\\Us
ers\\USER\\AppData\\Local\\Temp\\pip-install-ru5dt2hh\\arrex_198cb0a2b1904a24a0a5f2bce1f516f6\\setup.py'"'"';f=getat
tr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(comp
ile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\USER\AppData\Local\Temp\pip-pip-egg-info-57yeylj
w'
         cwd: C:\Users\USER\AppData\Local\Temp\pip-install-ru5dt2hh\arrex_198cb0a2b1904a24a0a5f2bce1f516f6\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
[... 50 lines cut ...]
eck the logs for full command output.
ERROR: Could not find a version that satisfies the requirement arrex>=0.3.1 (from pymadcad)
ERROR: No matching distribution found for arrex>=0.3.1
```

[... cut ...]
s://pypi.org/simple/pymadcad/) (requires-python:>=3.8). Requested pymadcad==0.13.1 from https://files.pythonhosted.o
rg/packages/42/2d/b972cda45e840c268548ee41f3d74bbd07aec3aaebe6eb77a98abb33999a/pymadcad-0.13.1.tar.gz#sha256=fe2727b
6e8b3a7c8364896666f31d6214211ad66eee6bdf826d00670096edf86 has inconsistent version: filename has '0.13.1', but metad
ata has '0.9.3'
ERROR: Could not find a version that satisfies the requirement pymadcad==0.13.1
ERROR: No matching distribution found for pymadcad==0.13.1
```

jimy-byerley, 2022-09-25:
That's a weird issue with for the first trace. I don't know what is the cause: I'm using python v3.8 on my side too. I guess this is an issue with the anaconda environment

For the second trace, it seems to reflect a warning that's showing on my side when I do the same. This is due to file that I packed by accident with the module (It was in the same folder ...). I just need rebuild the package properly.
I released v0.13.2 that should fix it

Eva-xx, 2022-09-25:
Great! Thank you for your reply！

closed 2022-09-25 by Eva-xx