jimy-byerley/pymadcad #52: note_angle acting weirdly

Yarkane, 2022-11-15:
Hey there !

I guess I'm not using the function correctly... But I can't manage to make this work. Sorry if this is a dumb question !

In the scheme module, you implemented many useful note functions including "note_angle" : https://pymadcad.readthedocs.io/en/latest/reference/scheme.html#madcad.scheme.note_angle

I try to use it to make the user create angles by clicking:

```
<At each click on the 3D scene with the angle notation tool:>
               if self.first_point is None:
                    self.first_point = self.ptat(pos)
                elif self.second_point is None:
                    self.second_point = self.ptat(pos)

                    # Display the "AB" segment
                    measure_keys = self.make_measure(self.first_point, self.second_point)
                else:
                    _third_point = self.ptat(pos)

                    # display the "BC" segment
                    measure_keys = self.make_measure(self.second_point, _third_point)
                    
                    # Compute the angle and display it
                    a1 = madcad.Axis(self.second_point, self.first_point)
                    a2 = madcad.Axis(self.second_point, _third_point)
                    note_angle = madcad.note_angle(a1, a2)
                    note_key = self.scene.add(note_angle)
```

The "note_angle" function renders me something like that :
[image]
[image]

I can't show it but it seems the measure indicates "81°" which is... not that far, but I am not sure about this value neither.

Do you have any tip about what I may be doing wrong ?

Thanks a lot (again) ! :)

Have a good day !

jimy-byerley, 2022-11-15:
Very weird display !  
I notices the following two lines intending to create `Axis` from couple of points, is it what you do in your real code ?
```python
a1 = madcad.Axis(self.second_point, self.first_point)
a2 = madcad.Axis(self.second_point, _third_point)
 ```
In such case that might be the source of the issue because `Axis` expects an origin point and a direction vector (which is a point in a 3d vector space mathematically speaking, but not a point on the axis geaometrically speaking). if you want to create an axis from two points `(a,b)` in your space, you should call 
```python
Axis(a, normalize(b-a))    # b-a  is the vector from a to b, we normalize it to get a pure direction (free of any notion of length)
```

It could also be a bug, but it's unlikely as it always worked well for me, as [`tests/test_scheme.py`](tests/test_scheme.py) shows
If it is a bug I need a concrete example to see what's wrong in this case ... since it works well when I try it

Yarkane, 2022-11-16:
Hi !

You are right, I was indeed using `madcad.Axis()` the wrong way. Thanks to your explanation I was able to make it work ! 
Thanks a lot for your time :)

closed 2022-11-16 by Yarkane