jimy-byerley/pymadcad #57: missing .gitignore,

GlennWSo, 2022-12-23:
Hi,

As developer and someone who wants to contribute to this project I am missing a ".gitignore" file.

Without the  .gitignore file i do not know which files should _not_ included in future pull requests.

jimy-byerley, 2022-12-23:
Hum ... isn't that simpler to only include the source files you created ?

GlennWSo, 2022-12-23:
Tools like "git status" has greatly improved signal to noise ratio with a .gitignore file. It also clearly communicates to contributors so they know what u don't want.

Merry Christmas btw 🌲

jimy-byerley, 2022-12-24:
That's right ... what afraid me is that there will always be new IDEs or tools creating their own temporary files with always different names. And the gitignore is always bigger and bigger. I use gitignore files without comiting them, so they are specific to the computer and tools I'm using them on.
Well I suppose it's not that important ;)

Merry Christmas for you too :christmas_tree:

PR #58 (2022-12-24): add .gitignore
Added git ignore using the python template from https://github.com/github/gitignore

resolves #57
closed 2022-12-24 by jimy-byerley