jimy-byerley/pymadcad #61: Mac OS m-1 install

triple7, 2023-01-03:
Hello,

I've been trying to get the module installed on a Mac OS m-1. I guess a few things here.

First, python is currently miniconda for me [when running which python, the top of the list is conda].

I tried installing using pip with a non condo bistro, and it has the following issue with PyQt5:
      AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_f

I tried updating pip to the latest and retried, no go. An article on the net shows us how to do this with a rosetta2 terminal session, still no go.

So I tried getting into the condo environment with:
Honda activate tf (tf being my environment) and ran pip install --upgrade pip

All my condo packages got upgraded, and in there PyQt5 is available.

Finally, running pip install pymadcad after having installed all dependencies, it's stuck at:
  Using cached PyQt5-5.15.7.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... -

It's been stuck there for over 10 minutes now so I expect it to be an issue.

Anyone successfully installed pyMADCAD on a Mac OS m-1?

jimy-byerley, 2023-01-03:
Hello @triple7 
If pip is using a `.tar.gz` package to install `PyQt5`, it means no precompiled package is available for your platform (mac os m-1 might no tbe mainstream) so `pip` is using the source archive `PyQt5-5.15.7.tar.gz` to compile PyQt5
It is them perfectly normal that the installation takes a while since Qt is big ... it can easily take 20min even on modern computers
No shure why however pip get stucks particularly on step "preparing metadata (pyproject.toml)"

I don't have experience in compiling such a big library **in a conda environment**. I don't know if all the compilation tools and pyQt dependencies are made available by conda. You might need to install some dependencies by yourself :-/

kzlar, 2023-10-05:
You're likely stuck on license confirmation. Try the following command:
`pip install pyqt5 --config-settings --confirm-license= --verbose`
and give it about an hour.

jimy-byerley, 2023-12-23:
No news on this issue in months, I am going to assume @kzlar solution worked and then close this issue

closed 2023-12-23 by jimy-byerley