jimy-byerley/pymadcad #69: Install from sources fails ( trying to get pymadcad into docker arm64 image )

manuel-koch, 2023-02-02:
I'm trying to get pymadcad into a docker image for arm64 ( Apple M1 ).
Unfortunately the plain `pip install pymadcad` doesn't work as PyQt5 is not provided as arm64 wheel and the build fails.

Afterwards I tried alternative instruction as suggested [here](https://pymadcad.readthedocs.io/en/latest/installation.html#compile-the-module).

```shell
$ git clone https://github.com/jimy-byerley/pymadcad.git
$ cd pymadcad
$ python3 setup.py build_ext --inplace       
running build_ext
building 'madcad.core' extension
creating build
creating build/temp.linux-aarch64-cpython-310
creating build/temp.linux-aarch64-cpython-310/madcad
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/usr/local/include/python3.10 -c madcad/core.c -o build/temp.linux-aarch64-cpython-310/madcad/core.o
gcc: error: madcad/core.c: No such file or directory
gcc: fatal error: no input files
compilation terminated.
error: command '/usr/bin/gcc' failed with exit code 1
```

Looking at the cloned repo, there is indeed no file `madcad/core.c`.

What am I missing ?

GlennWSo, 2023-02-02:
the file: madcad/core.c should be generated by cython.

as workaround u could try add:
> cython madcad/core.pyx
just before calling setup.py

GlennWSo, 2023-02-02:
> Is it possible to easily strip any display/view stuff from pymadcad ?
No, i dont think there is a very convenient way of doing this.

Currently in madcad.__init__.py at line number 167 
```
from .rendering import Scene, Display, displayable, show, render
```
and rendering intern imports lots pyqt stuff.

So u have do some work to use pymadcad without pyqt. 

I suggest fork madcad or atleast create new issue "make madcad.show() an optional feature"

GlennWSo, 2023-02-02:
another idea is to create a dummy pyqt.
that have all these things:
from PyQt5.QtCore import Qt, QPoint, QEvent
from PyQt5.QtWidgets import QApplication, QWidget, QOpenGLWidget
from PyQt5.QtGui import QSurfaceFormat, QMouseEvent, QInputEvent, QKeyEvent, QTouchEvent, 

like:
```
def QEvent():
    pass
```

but if honstly the quickest way would be just fork pymadcad

manuel-koch, 2023-02-02:
Thanks for the quick reply! Maybe I'll try the fork suggestion.

You can close this issue if you like.

jimy-byerley, 2023-02-03:
I confirm: madcad cannot at the moment be imported without pyqt installed. However It wouldn't ask too much work to acheive. 

- the rendering pipeline itself only depends on openGL, and really needs it working only when instantiated
- the `View` widget has a base class `ViewCommon` that does not depend on Qt either

I think it could be a nice addition to make pyqt an optional dependency, since most of the madcad features are just computational. But that's something we cannot just hack at the moment since it raises many question we have to answer first before refactoring the module.

- how to choose which rendering backend (if any) is used when madcad is imported ?
- should we move to PySide6 (Qt6) ?
- should we support both PySide2 and PySide6 (Qt5 and Qt6) ?
- `show()` shall stay in the root module (for convenience), so what is the best way to make it initialize the Qt backend if it is not already initialized ?
- what *practices* should we enforce in madcad to let the madcad objects declare their own `.display()` method and classes without importing Qt

This case has to be started someday I think. If we are about to do it, we should open a new issue ;)
Thanks @GlennWSo for having answered that topic while I was busy !

closed 2023-02-03 by jimy-byerley