jimy-byerley/pymadcad #70: show([mesh]) mutates mesh

GlennWSo, 2023-02-02:
hi

i expected that show([mesh]) should not mutate mesh.

This causes problems when debugging, I cant inspect visually without mutating my objects.

simple snippet:
```
from madcad import icosahedron, show

ico = icosahedron(vec3(0), 1)
print(ico)
# > Mesh with 12 points, 20 faces
show([ico])
print(ico)
# > Mesh with 62 points, 20 faces
```

jimy-byerley, 2023-02-03:
Hi !
That's ... odd indeed. I suspect this is because of some extraction methods of the mesh being called in `Mesh.display()` for getting the group edges, wireframe, etc

jimy-byerley, 2023-02-03:
I got it:  `Mesh.display()` is duplicating some points to have separate vertices in the openGL buffer. Those duplication could happen on a copy of the original buffer instead.
It is much better to fix that because a display is like a measurement: it shall not affect what we measure.

closed 2023-02-10 by jimy-byerley