jimy-byerley/pymadcad #71: make pyqt optional

GlennWSo, 2023-02-03:
Discussion in #69 reveled that having pyqt as a optional dependency would be valuable for some users

> I confirm: madcad cannot at the moment be imported without pyqt installed. However It wouldn't ask too much work to acheive.
> 
>     * the rendering pipeline itself only depends on openGL, and really needs it working only when instantiated
> 
>     * the `View` widget has a base class `ViewCommon` that does not depend on Qt either
> 
> 
> I think it could be a nice addition to make pyqt an optional dependency, since most of the madcad features are just computational. But that's something we cannot just hack at the moment since it raises many question we have to answer first before refactoring the module.
> 
>     * how to choose which rendering backend (if any) is used when madcad is imported ?
> 
>     * should we move to PySide6 (Qt6) ?
> 
>     * should we support both PySide2 and PySide6 (Qt5 and Qt6) ?
> 
>     * `show()` shall stay in the root module (for convenience), so what is the best way to make it initialize the Qt backend if it is not already initialized ?
> 
>     * what _practices_ should we enforce in madcad to let the madcad objects declare their own `.display()` method and classes without importing Qt
> 
> 
> This case has to be started someday I think. If we are about to do it, we should open a new issue ;) Thanks @GlennWSo for having answered that topic while I was busy !

jimy-byerley, 2023-03-12:
This feature is added in [the plan](https://github.com/jimy-byerley/pymadcad-refactor-rendering) to improve the rendering pipeline

jimy-byerley, 2023-04-11:
@bourbonut  is working on a refactoring of the rendering pipeline that could address this feature, It will follow the architecture discussed on [this repo](https://github.com/jimy-byerley/pymadcad-refactor-rendering)

jimy-byerley, 2025-05-01:
it is now implemented

closed 2025-05-01 by jimy-byerley