jimy-byerley/pymadcad #73: bug: boolean.cut_mesh

GlennWSo, 2023-02-23:
## problem
the returned forntier: Web sometimes have missing edges
[image]

## example:
```python
from madcad import *

cut_tool = icosphere(vec3(0), 1, resolution=("div", 1))
plane = square(Axis(vec3(0), Z), width=4)
mesh, web = boolean.cut_mesh(plane, cut_tool)

show([web])
```
I expected that example snippet would show a web that forms a closed loop. It does not

jimy-byerley, 2023-04-11:
Strange it does not form a closed loop. I will investigate

jimy-byerley, 2023-05-22:
I found the cause of the issue, a stupid mistake I made in `core.intersect_triangles`
I will write a non-regression-test for this

I'm surprised it did not caused more troubles in the boolean operations until now, that's fun you noticed it on a wrong result of `cut_mesh()` rather than on a boolean operation on a bigger mesh.
Anyway, thanks for reporting it !

commit 323cd9ff (2023-05-22): fixed #73
closed 2023-05-22 by jimy-byerley