jimy-byerley/pymadcad #75: Cannot del version   from mathutils,  and  AttributeError: module 'glm' has no attribute 'u8vec2'

ejlmljt, 2023-04-19:
pymadcad 0.15.1, Python 3.9.7 , Windows 10, install "pip install pymadcad "Feb 19,2023
Ran from windows cmd  window  "python  madcad_test1.py"

I had to change mathutils.py when script errored out with unfound module 
Change  from 

del version, license  

to

try: 
	del version, license  
except:
	print ("Could not delete module version,license in  mathutils") 

Also 
AttributeError: module 'glm' has no attribute 'u8vec2'

Traceback (most recent call last):
  File "C:\Users\XXXX\OneDrive\Documents\madcad\madcad_test1.py", line 3, in <module>
    from madcad import *
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python39\lib\site-packages\madcad\__init__.py", line 135, in <module>
    from . import (
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python39\lib\site-packages\madcad\mathutils.py", line 21, in <module>
    import arrex.glm
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python39\lib\site-packages\arrex\glm.py", line 28, in <module>
    type = getattr(glm, prec+'vec'+str(size))
AttributeError: module 'glm' has no attribute 'u8vec2'

Thanks

jimy-byerley, 2023-04-21:
Weird issue, module `glm` usually has a class `u8vec2`
Can you please try to run the following in a python shell ?

```python
import glm
print(glm)
print(glm.version)
```

ejlmljt, 2023-04-24:
Thanks   for the  reply  Jim,

I uninstalled the  python glm package that was installed with "pip
uninstall glm" and  pymadcad now  works!!
Error  was caused by conflicting glm modules

It just  took  seeing the word 'weird' in your reply to think it was
something unique to my computer.

Here is  my quick check on the installation as it  was when pymadcad  was
giving the error.

import glm
print (glm.version())

v0.4.4

>>> import os.path
>>> print(os.path.abspath(glm.__file__))

C:\Users\XXX\AppData\Local\Programs\Python\Python39\lib\site-packages\glm\__init__.py

I  looked up glm in  pypi.org and the package is a fast GridLAB-D to JSON
(and back) parser. I don't  see the connection from my cursory look  at
pymadcad.
So I  deleted glm  with

pip uninstall glm

Futher info on  the  glm package that was  installed.
glm 0.4.4
https://pypi.org/project/glm/
glm PyPI Python version compatibility status Build Status
A fast GridLAB-D to JSON (and back) parser.

Fast GLM to JSON converter.
Supports nested objects, schedules, definitions, directives, modules,
includes and clock objects
Fast JSON to GLM converter.
Native support for Python, returns nested dictionaries

jimy-byerley, 2023-04-24:
That's right, the correct glm module for `pymadcad` is [`pyglm`](https://pypi.org/project/PyGLM/) and not `glm` from [here](https://pypi.org/project/glm/)
They are named likewise once installed, but usually that's not a problem when leaving the dependencies resolution to pip ...  still this should be harmonized.

Well I'm glad you solved your issue. Come back if you have any other issue ;)

closed 2023-04-24 by jimy-byerley

ejlmljt, 2023-04-24:
Thanks a lot.