jimy-byerley/pymadcad #86: chamfer question

xiaodongdong101, 2023-07-25:
If I input a more complex triangular mesh model, can I still chamfer the inner edges?
[image]

jimy-byerley, 2023-07-29:
Sure, `chamfer` and `bevel` functions are designed to cut not only adjacent triangles to the given edge, but to go through all the mesh triangles to acheive the desired cut surface.

The limitation to it is the presence of crossover edges in the mesh, the current chamfer algorithm cannot deal with crossover.
chamfering flat edges may also not work depending of the `cutter` parameter you choose.

xiaodongdong101, 2023-08-01:
Could you help me test this model?
I haven't configured your code environment yet
[cop_4_of_clip.zip](https://github.com/jimy-byerley/pymadcad/files/12224001/cop_4_of_clip.zip)

jimy-byerley, 2023-08-07:
I can but what operation do you want to apply on it ? what is your problem with that mesh ?

xiaodongdong101, 2023-08-08:
Thanks a lot.Can you make bevel along a curved line? I want to see if it can handle a more special environment

xiaodongdong101, 2023-08-09:
Hello,I find that chamfer and bevel function are used in a straight edge.
[image]
Can this algorithm be used for bevel and chamfers of curves?
[image]

jimy-byerley, 2023-08-10:
Sure it can, `tangentend` is internally used only for each extremities of such curves

Here is a chamfer on the lower outer edge
[image]

```python
part = read('/tmp/cop_4_of_clip.stl')
part.mergeclose()
part = reverse.segmentation(part, tolerance=inf)
part.mergeclose()
part.check()
chamfer(part, part.frontiers(0, 1), ('width', 1))
```

My algorithm fails on the upper outer edge, I will have to inverstigate

xiaodongdong101, 2023-08-10:
Thanks a lot.Looking forward to your good news

xiaodongdong101, 2023-08-11:
If i want to choose any edge to bevel, how should I do it?
Thanks a lot

jimy-byerley, 2023-08-11:
Just like I wrote in my example, you have to provide a `list` of edges or a `Web` (by definition it contains a list of edges) for the edges you want to chamfer
In my previous code sample I wrote `part.frontiers(0, 1)` to select the edges at the frontier between groups `0` and `1`

Since your part is imported, it does not carry surface groups, so I added the segmentation step to automatically produce surface groups based on the surface curvature

```python
# import your mesh, without surface groups and without connectivity
part = read('/tmp/cop_4_of_clip.stl')
# a STL file does not have connectivity informations, so here we merge similar points and connect triangles together, chamfering will then become possible
part.mergeclose()
# produce surface groups out of surface curvature, edge selection will then become possible
part = reverse.segmentation(part, tolerance=inf)
# execute chamfer
chamfer(part, part.frontiers(0, 1), ('width', 1))
```

xiaodongdong101, 2023-08-12:
Thank you very much for your reply.But I can't test other simple grid models, and I don't know if there is something wrong with my parameters.
[image]
Here is what he reported wrongly
[image]
This is the model that I input
[image]
Here is my input data
[test0812.zip](https://github.com/jimy-byerley/pymadcad/files/12326254/test0812.zip)

xiaodongdong101, 2023-08-12:
Why do I use the subdivided mesh model as input, but the result has no effect.
[image]

[cop_4_of_clip_1.zip](https://github.com/jimy-byerley/pymadcad/files/12326718/cop_4_of_clip_1.zip)

jimy-byerley, 2023-08-12:
I think there is an issue with subdivided meshes, because it makes points in the middle of straight lines on the chamfered edges

xiaodongdong101, 2023-08-15:
Is there any better solution?Because the fine mesh model also has a complete topology,How should we solve this problem is better?

xiaodongdong101, 2023-08-16:
Hello,Why is the offset in the loop assigned multiple times?
**********************************
for b in prox:
			for c in prox:
				a = junc
				if b == c or b == a or c == a:	continue
				# calculer la distance
				ab = normalize(pts[b]-pts[a])
				ac = normalize(pts[c]-pts[a])
				n = cross(ab, ac)
				nb = cross(n, ab)
				nc = cross(ac, n)
				ob = offsets[edgekey(a,b)]
				oc = offsets[edgekey(a,c)]
				ib = dot(ob,ob) / dot(ob, nb)
				ic = dot(oc,oc) / dot(oc, nc)
				s = dot(nb,ac)
				if not s:	continue
				abl = (ib*dot(ab,ac) + ic) / s
				offset = dot(abl*ab + ib*nb, normals[a])
**********************************
So how to save during the loop?
https://pymadcad.readthedocs.io/en/latest/_modules/madcad/cut.html#bevel
Does this offset value need to be accumulated multiple times?
*********************************************
[image]

jimy-byerley, 2023-08-28:
> Is there any better solution?Because the fine mesh model also has a complete topology,How should we solve this problem is better?

A better solution would be a better implementation in this case, this is a bug I should work on ;) but I need to find time for it.

> Why is the offset in the loop assigned multiple times?

Don't forget to format the code you post in python code blocks, just [as in regular makdown](https://www.markdownguide.org/extended-syntax/#syntax-highlighting)

I guess you are speaking of [this line](https://github.com/jimy-byerley/pymadcad/blob/38463deca2011e452d716e5c9581318a8d5b81ce/madcad/cut.py#L367) (don't forget just a code screenshot doesn't help much to understand where it is in the library).

I don't remember why I did this that way. you are right, maybe some accumulator is missing here, something like a `min`