jimy-byerley/pymadcad #87: Issue with 'difference' Function - Incorrect Object Removal and Polygon Closure

Tommamgo, 2023-08-28:
Hello,

I hope you're doing well. I've come across an issue while testing the 'difference' function. During the process of subtracting an STL file from another, I observed that while the object's orientation was correctly subtracted, the polygons were not closed properly.

## Code: 
```
from madcad import *

#open
read_mesh1 = io.read("stl1.stl")
read_mesh2 = io.read("stl2.stl")
read_mesh1.mergeclose()
read_mesh2.mergeclose()

# Remove the volume of the second to the first
diff = difference(read_mesh2, read_mesh1)

# Write
io.write(diff, "mesh.stl")
```

# Result: 
[image]

# This should be the Result: 
[image]

## stl1 and stl2: 
[stl.zip](https://github.com/jimy-byerley/pymadcad/files/12454863/stl.zip)

jimy-byerley, 2023-08-28:
I cannot reproduce the issue on my side with the master branch
Which version of pymadcad do you use ?

jimy-byerley, 2023-08-31:
This discussion should be a different issue then

I don't know exactly what is `stl1.stl` and `stl2.stl` in your example since your zip file only contains `Basis.stl` and `Gemin.stl`
However that seems to be a floating point precision issue: I would avoid the problem with a slight shift of one of your meshes

```python
difference(basis, gemein.transform(0.001*vec3(1)))
```

Tommamgo, 2023-09-01:
Thank you for the quick response.

I apologize for the incorrect file naming.

stl1.stl = Basis.stl
stl2.stl = Gemin.stl

With your changes, I was able to compile it and achieved the desired results.

Thanks!

jimy-byerley, 2023-09-02:
Happy that it worked for you !
I will keep this issue opened since the floating point precision issue is a bug that needs to be solved
Thanks for reporting