jimy-byerley/pymadcad #88: problem with numpy >1.24.4

EdwinSoft, 2023-09-10:
When I try the following code with numpy 1.24.4, it works fine
```
from madcad import brick
from madcad.rendering import *

import sys
from PyQt5.QtWidgets import QApplication

scn = Scene({
    'brick': brick(Box(width=vec3(1))),
})

app = QApplication(sys.argv)
settings.use_qt_colors()
view = View(scn)
view.resize(800, 800)
view.show()
view.look(fvec3(-1, 0.15, -1))
interest = view.scene.box()
view.center(interest.center)
view.adjust(interest)
sys.exit(app.exec())
```

[image]

but with a higher version of numpy it generates an error and does not show anything (tested with numpy 1.25.2 and 1.26.0rc1).
[image]
```
tried to display <madcad.mesh.mesh.Mesh object at 0x7f49a8089df0>
Traceback (most recent call last):
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/madcad/rendering.py", line 516, in dequeue
    self.displays[key] = self.display(displayable, self.displays.get(key))
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/madcad/rendering.py", line 598, in display
    disp = obj.display(self)
           ^^^^^^^^^^^^^^^^^
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/madcad/mesh/mesh.py", line 663, in display
    return displays.SolidDisplay(scene,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/madcad/displays.py", line 259, in __init__
    self.disp_faces = FacesDisplay(scene, self.vertices, normals, faces, color=color, reflect=reflect, layer=0)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/madcad/displays.py", line 389, in __init__
    self.vb_faces = scene.ctx.buffer(np.array(faces, 'u4', copy=False))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/moderngl/__init__.py", line 1732, in buffer
    res.mglo, res._size, res._glo = self.mglo.buffer(data, reserve, dynamic)
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: ndarray is not C-contiguous
Exception ignored in: <function FacesDisplay.__del__ at 0x7f49b41c8180>
Traceback (most recent call last):
  File "/home/edwincordoba/PycharmProjects/test_pymadcad/venv/lib64/python3.11/site-packages/madcad/displays.py", line 409, in __del__
    if self.va:
       ^^^^^^^
AttributeError: 'FacesDisplay' object has no attribute 'va'
```
os: Fedora 38
Python 3.11.4

Pillow	10.0.0	
PyGLM	2.7.0	
PyQt5	5.15.9	
PyQt5-Qt5	5.15.2	
PyQt5-sip	12.12.2	1
PyYAML	6.0.1	
arrex	0.5.2	
glcontext	2.4.0	
moderngl	5.8.2	
pip	23.2.1	
pymadcad	0.15.1	
scipy	1.11.2	
setuptools	68.2.0	
wheel	0.41.2

jimy-byerley, 2023-09-12:
Hello This looks like a duplicate of #85 which is solved on the master branch (ibut not released yet), can you try it ?

EdwinSoft, 2023-09-12:
Thanks Jimy. I tried it with master branch and it solves the problem.

closed 2023-09-12 by EdwinSoft