jimy-byerley/pymadcad #89: AttributeError: 'FacesDisplay' object has no attribute 'va'

1939938853, 2023-09-13:
Hello,

I just run the demo code

`
m = screw(10, 20)
show([m]
`
and got  the AttributeError. Below is the detailed information. Thanks

`
tried to display <madcad.mesh.mesh.Mesh object at 0x000001D2AE97F220>
Traceback (most recent call last):
  File "D:\Anaconda3\envs\sp1\lib\site-packages\madcad\rendering.py", line 675, in update
    self.displays[key] = disp = scene.display(obj, self.displays.get(key))
  File "D:\Anaconda3\envs\sp1\lib\site-packages\madcad\rendering.py", line 598, in display
    disp = obj.display(self)
  File "D:\Anaconda3\envs\sp1\lib\site-packages\madcad\mesh\mesh.py", line 663, in display
    return displays.SolidDisplay(scene,
  File "D:\Anaconda3\envs\sp1\lib\site-packages\madcad\displays.py", line 259, in __init__
    self.disp_faces = FacesDisplay(scene, self.vertices, normals, faces, color=color, reflect=reflect, layer=0)
  File "D:\Anaconda3\envs\sp1\lib\site-packages\madcad\displays.py", line 389, in __init__
    self.vb_faces = scene.ctx.buffer(np.array(faces, 'u4', copy=False))
  File "D:\Anaconda3\envs\sp1\lib\site-packages\moderngl\__init__.py", line 1732, in buffer
    res.mglo, res._size, res._glo = self.mglo.buffer(data, reserve, dynamic)
ValueError: ndarray is not C-contiguous
Exception ignored in: <function FacesDisplay.__del__ at 0x000001D297DFF4C0>
Traceback (most recent call last):
  File "D:\Anaconda3\envs\sp1\lib\site-packages\madcad\displays.py", line 409, in __del__
    if self.va:
AttributeError: 'FacesDisplay' object has no attribute 'va'

`

jimy-byerley, 2023-09-13:
I think I will make a release for this fix, so no one will get trapped into this anymore

1939938853, 2023-09-18:
I tried to install it from the master branch, but got file madcad/core.c cannot be fund error.

`
(sp1) D:\git\mesh\pymadcad>python setup.py build_ext --inplace
running build_ext
building 'madcad.core' extension
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -ID:\Anaconda3\envs\sp1\include -ID:\Anaconda3\envs\sp1\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /Tcmadcad/core.c /Fobuild\temp.win-amd64-cpython-39\Release\madcad/core.obj
core.c
c1: fatal error C1083: Cannot open source file: 'madcad/core.c': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

(sp1) D:\git\mesh\pymadcad>
`

EdwinSoft, 2023-09-18:
> I tried to install it from the master branch, but got file madcad/core.c cannot be fund error.
>
> ` (sp1) D:\git\mesh\pymadcad>python setup.py build_ext --inplace running build_ext building [...]

Did you install Cython?

1939938853, 2023-09-18:
Yes, after installing Cython, I was able to to install the mater branch version pymadcad.

Thank you EdwinSoft

jimy-byerley, 2023-09-18:
Thanks @EdwinSoft :+1:
 If there is no problem related to the backtrace in the initial post, I will close this issue.

closed 2023-09-18 by jimy-byerley