jimy-byerley/pymadcad #92: Is it possible to bevel the "unknown" sharp edge generated by the thicken operation

1939938853, 2023-09-27:
Dear Jimy,

Is it possible to bevel the "unknown" sharp edge generated by the thicken operation? I saw Blender can set roundness here in Thicken operation. For Madcad perhaps need to extract all original edge points first and then apply Madcad Bevel operation?

jimy-byerley, 2023-09-29:
Hello
What are these "unknown" edges ?
As the [documentations says](https://pymadcad.readthedocs.io/en/latest/reference/cut.html#madcad.cut.bevel), it is indeed needed to provide the set of edges you want to bevel. Do you mean you don't know how to extract such a set of the edges you want to bevel ?

1939938853, 2023-09-29:
Hello Jimy,

Yes. Is there any function in pymadcad that can be used to extract edges, like the one in[ pyvista](https://docs.pyvista.org/version/stable/examples/01-filter/extract-edges.html)?

[image]

jimy-byerley, 2023-09-29:
Yes there is in fact 2 steps for doing so:
- segment the surface into groups with [`madcad.reverse.segmentation()`]() to, so edges will appear to distinguish groups (groups are regular portions of surface)
- select the desired group frontiers, there is 2 ways of doing so:
  + 1: using `Mesh.frontiers()` in combinations with `Mesh.groupnear()`
  + 2: using a progagative selection to pick part of frontiers based on their topology
     the problem here is that it is not documented yet, because I never polished this API :sweat:  The function is [`madcad.selection.select()`](https://github.com/jimy-byerley/pymadcad/blob/4f7c97c57240fcc51e14d7a3f961d9cfa6b97cbf/madcad/selection.py#L13)

Note that the first step is in case you do not have surface grouping at all, if your mesh comes from `madcad.extrusion()`, it has already groups defined

Option 1 could look like this:
```python
raw_part = ...  # mesh without group informations
part = mc.reverse.segmentation(raw_part)
# suppose we want to select the edges arround the top face and a side face
edges = part.frontiers(
    part.groupnear(vec3(0,0,1)),   # assuming this point is somewhere above the mesh
    part.groupnear(vec3(2,0,0)),  # asuming this point is somewhere next to the mesh
    )
# then do whatever you want with these edges
bevel(part, edges, ('width', 0.1))
```

1939938853, 2023-09-30:
Thank you, Jimy!

jimy-byerley, 2023-10-04:
Are you sure this mesh is an output from `thinken` on a madcad-generated mesh ?
It looks like either - you are thinkening an imported soup of triangles (like a stl file) - or you used an other operation than thinkening

1939938853, 2023-10-05:
Hi Jimy,

It is true that the mesh is imported from a stl file. Can you give me some suggestions on how to make an organized mesh from soup triangles? Thank you!

jimy-byerley, 2023-10-05:
> Can you give me some suggestions on how to make an organized mesh from soup triangles?
In a stl file, the triangles points are supposed to coincide with neighboring triangles, so you just need to merge the close points together to obtain a connex mesh
```python
part = read('myfile.stl')
part.mergeclose()
```
Then you can apply the operations you want, like any madcad-generated mesh. While thinkening, you will see that the border are much cleaner ;)

1939938853, 2023-10-06:
Hi,

It is true  that `mergeclose()` make the thicken edges much better. But I got unwanted stuff after `thicken()` . Here is my stl file and below is my code. Any ideas?

Thank you!

`from madcad import *`
`from madcad.generation import *`
`mesh = read("mesh.stl")`
`mesh.mergeclose()`
`thickness = 4`
`thicken_mesh = thicken(mesh, thickness, 0.5)`
`show([thicken_mesh])`

[image]

[mesh.zip](https://github.com/jimy-byerley/pymadcad/files/12829709/mesh.zip)

closed 2023-10-07 by jimy-byerley