jimy-byerley/pymadcad #94: Question: Merging webs/meshes

kzlar, 2023-10-06:
Hi! Appreciate your effort on this library, it's really cool :)
I have a small question: Is there a way to merge webs (or meshes)?
Just as a super simple example, let's say I have two rectangles and I want to merge them into a single "L" shape mesh so I'd have a single closed loop instead of two. 
I searched the docs but the merge functions seem to deal with groups and not with other webs/meshes, but it's also possible that I misunderstood :)
Hope the question makes sense and that there's a way to do it 🤞

jimy-byerley, 2023-10-07:
Hello, glad you like it !
When your meshes/webs to merge shall not be in contact one with an other, you can use the concatenation operator `mesh1 + mesh2`. It is of course defined between meshes and between webs, but is non-sens between a mesh and a web
If however you need their geometries to be intersected to form a new shape, what you need is a [boolean operation](https://pymadcad.readthedocs.io/en/latest/reference/boolean.html), and of course it works between meshes and between webs (and some are sometimes possible between webs and meshes)

kzlar, 2023-10-09:
Ah yes, the union operation was what I was missing. I did have some peculiarities with union-ing webs that have been mirrored using `scaledir`. I needed to call `flip` on them, but it turned out to be a rabbit hole 😅 .
I ended up re-writing my code to remove the common edges and then the `+` operator worked.
Thank you for the pointers :)

I'll go ahead and close this as my initial issue is resolved. But if I may, I wonder if you have a recommendation for a python library that helps simulate deformation of parts that I create with pymadcad. Something like, `if I make this part from material X, and apply force F at point A, how would my part look?`

closed 2023-10-09 by kzlar

jimy-byerley, 2023-10-11:
> I wonder if you have a recommendation for a python library that helps simulate deformation of parts that I create with pymadcad. Something like, if I make this part from material X, and apply force F at point A, how would my part look?

I do not have much to recommend, I already used FreeFEM by the past, but it is not a python library at all, I used it in pure C++ and I do not know if some python wrapper emerged since.
I do not know python libraries for this, but there surely are some.