jimy-byerley/pymadcad #95: Thicken for a free-form mesh from stl file

1939938853, 2023-10-07:
Hi,

I tried to do the `thicken()` operation with a free-form mesh stored in a stl file. It seems not working. Any suggestions?

Thanks

Here is my code
```
from madcad import *
from madcad.generation import *
mesh = read("mesh.stl")
mesh.mergeclose()
thickness = 4
thicken_mesh = thicken(mesh, thickness, 0.5)
show([thicken_mesh])
```
Mesh before `thicken()`
[image]
Mesh after `thicken()`
[image]
The zipped stl mesh file

[mesh.zip](https://github.com/jimy-byerley/pymadcad/files/12837276/mesh.zip)

jimy-byerley, 2023-10-07:
Hello
This looks like a wrong thickness was computed for some points of your meshes, this can happen when the input meshes contains geometry artifacts like it is in your case.

Here is your input mesh
[image]
Below is a zoom on the artifact at the bottom of previous image
[image]

You can see that there is two triangles with 360° angle between them, which makes of it a paradoxical surface.
The wrong result is predictable:
[image]

If you fix that meshing artifact on your input mesh, your issue is solved

1939938853, 2023-10-08:
Thank you Jimy,

Below is the original mesh. Do you mean the triangle marked green has  360° angle with the triangle marked blue? How to judge this is 360° angle and not 0°?   I googled the "paradoxical surface", but cannot find a definition. What it is really about? It is true after `mesh.mergeclose()`, this part looks quite different.

[image]

jimy-byerley, 2023-10-11:
Yes I meant the green and the blue triangles are overlapping with opposite outward directions. 

In fact 0° is the same as 360° speaking of an orientation, in this case the vertex offset computed then tends to $\pm \infty$ because it cannot decide which side is interior and which is exterior.

I do not find a place defining what I called a paradoxical surface either, but let say that it is a surface going through itself, like a [Klein bottle](https://en.wikipedia.org/wiki/Klein_bottle)

jimy-byerley, 2023-10-11:
Im looking closely to your mesh, and I realize I was wrong, this is not a 360° problem at the position I pointed but a [winding](https://en.wikipedia.org/wiki/Winding_number) consistency issue in your triangles ... how did you generated this mesh, this is not very common that 3d softwares export meshes without widing consistency ?
If you flip the widing (the order of points in the triplet) of the blue triangle, then it is fine. 
Without this, the blue triangle is not overlapping with the green triangle, but yet has a normal opposite to the green one. We can see it by the blue triangle being darker than the green one.

[image]
[image]

recomputing winding consistency worked for the aforementioned position:
```python
mesh = read('/tmp/mesh.stl')
mesh.mergeclose()
mesh.orient()  # correct winding consistency of all triangles
result = thicken(mesh, 0.1)
```

But there is an other place in your mesh where the 360° angle arise
[image]
[image]

You can see that the triangles are indeed overlapping, it is not exactly 360° but close to it, so the vertex offset is not infinity but still huge:
[image]

jimy-byerley, 2023-12-23:
This topic has been inactive for few months, and since I explained this was not an issue with the thicken function but with the input mesh, I'm gonna close this issue

Feel free to reopen if you have more questions

closed 2023-12-23 by jimy-byerley