jimy-byerley/pymadcad #120: Issue with NumPy version 2.0

chryswoehler, 2024-10-02:
Hi, 

When I was trying to set up the repo to work on issue #115, found issue with version listed for the numpy library in requirements.txt. File currently lists version as `numpy>=1.1`

Downgrading to a version lower then 2.0 resolved the building error when trying to run the compile command:

"If using `np.array(obj, copy=False)` replace it with `np.asarray(obj)` to allow a copy when needed (no behavior change in NumPy 1.x).
For more details, see https://numpy.org/devdocs/numpy_2_0_migration_guide.html#adapting-to-changes-in-the-copy-keyword."

So it seems that the library has had a change to the copy keyword in it's 2.0 version.

Changing the version restriction for numpy to `numpy>=1.1,<=2.0` should solve the issue? Or would it be better to work towards updating for 2.0 keyword support rather then restricting to pre 2.0 library versions? 

This is my first open source contribution so I apologize if I overlooked something when setting up and this isn't an actual issue :)

chryswoehler, 2024-10-02:
This is my first time working on an open source project, sorry if this is not an actual issue :)

jimy-byerley, 2024-10-02:
Hello @chryswoehler, this totally seems to be an issue. The concept of issue is not very specific anyway. They serve as bug reports as well as feature request anyway ;)

changing the version requirement would be an easyfix of course but it is far more interesting to extend support to numpy 2 as you said. Since it doesn't seem much more than changing few numpy calls we should go for it.

chryswoehler, 2024-10-02:
I'll go ahead and just work towards numpy 2 support then. Thanks!

Oakchris1955, 2024-11-10:
Since 0.17 is now broken, shouldn't it be yanked?

jimy-byerley, 2024-11-13:
Yes old oversions a well, in fact numpy 2 was not existing while the previous madcad versions were released

jimy-byerley, 2024-11-13:
fixed in the new release

closed 2024-11-13 by jimy-byerley