jimy-byerley/pymadcad #125: Unable to pip install on Windows

harishsk, 2024-12-10:
Environment:
 - Windows 11
 - Python 3.12 in virtual environment created by Mamba
 - Visual Studio 2022 Community Edition
 - MSVC Tools 14.42.34433
 - CL version 19.42.34435

I am getting 177 linker errors in from core.obj and 153 linker errors from dtypes.obj. They are all of the format:

```
dtypes.obj : error LNK2001: unresolved external symbol __imp__PyBaseObject_Type
dtypes.obj : error LNK2001: unresolved external symbol __imp__PyGC_Enable
```

These functions are in python312.lib but they are defined with a single underscore (e.g. `__imp_PyBaseObject_Type`) instead of the one with double underscore the generated object file is looking for. 

Any idea how to resolve this? 

Are there precompiled binaries available for Windows?

Thanks.

jimy-byerley, 2024-12-10:
This looks like a Cython issue
Which version of cython do you use ?

harishsk, 2024-12-11:
`cython.exe --version` = `Cython version 3.0.11`

jimy-byerley, 2024-12-13:
Well since it is a linker issue and nothing in eve_core is refering explicitely to the compiled objects mentionned in this error, i suspect it is an issue in the translation from cython code into c code ...
I think i will try to reproduce youe issue this week end on a win11 machine, so I should be anyway able to provide you a precompiled package this time

harishsk, 2024-12-30:
I looked into this some more. I have two theories. One is that it is some issue related to my Visual Studio environment, but I can't figure out what that might be. The other possibility is that pymadcad is being built for 32bit but my Python version is 64bit.

Any thoughts? Will there be a precompiled package for 64-bit Python?
Thanks!

jimy-byerley, 2024-12-30:
On my side I tried to install the environment to build it on a computer of mine, but something went wrong in the install and `msvc` was not found by the `setup.py` script  ... I have to retry the next time I have this computer under the hands

harishsk, 2024-12-31:
Sorry, user error. I had used VsDevCmd.bat to initialize the Visual Studio tools and it turns out that it initializes it with x86 tools. To use the x64 compiler it is necessary to use vcvars64.bat. It installs now. Thanks for looking into it.

closed 2024-12-31 by harishsk

jimy-byerley, 2024-12-31:
Hum I find odd that these symbol errors came from an arch issue, but glad you found it !