jimy-byerley/pymadcad #131: Open letter about the status of pymadcad

Oakchris1955, 2025-03-15:
https://gist.github.com/Oakchris1955/47293d1a3f0c4cd42ab7c74a4af4635c

jimy-byerley, 2025-03-20:
Hello !
Thanks for this feedback, it is good for me to know what people who have used madcad for a while do think about it.

regarding the issues comming from dependencies, I must say you are right about it being anoying. Maybe we could solve this by setting up automated tests against many environments and many dependencies versions. This would certainly help detecting compatiblity issues instead of relying on the community to report it.
The compatiblities fix may however still take some time, as it is a one-man project ATM and there is moment I do not have time to work on pymadcad

regarding the documentation, It is good you say it to me. In fact for the meaning of `Axis`  and the mesh concatenation, I used it so many times that It appears completely obvious to me. I understand it might not be.
I will have to improve the docs in this matter

Do you think it should appear as more examples and tutorials in the docs ? or do you think this should be stated somewhere in the documentation of `Circle` or `Axis` ?

Oakchris1955, 2025-03-22:
> The compatiblities fix may however still take some time, as it is a one-man project ATM and there is moment I do not have time to work on pymadcad

I am on a similar situation myself, lol.

> regarding the issues comming from dependencies [...] Maybe we could solve this by setting up automated tests against many environments and many dependencies versions.

I believe that a weekly cron CI that runs on the latest Python version could help with that.

> regarding the documentation [..] for the meaning of Axis and the mesh concatenation [..] Do you think it should appear as more examples and tutorials in the docs ? or do you think this should be stated somewhere in the documentation of Circle or Axis ?

Perhaps both, but I would start with the latter.

I would love to help with all these issue, but right now I got my own projects running in parallel. When I find free time, I will make sure to open a pull request