jimy-byerley/pymadcad #134: switch packaging to `poetry`

jimy-byerley, 2025-06-07:
currently we are using
- `setuptools` to compile and package
- `twine` to distribute

the first is deprecated for packaging and the second is yet an other tool
`poetry` has become more standard-compliant and easy over time, so lets user it

We will need to have two files
- `pyproject.toml` using poetry
- `build.py` for cython compilation

nahaktarun, 2025-06-07:
Hi @jimy-byerley  I'm interested in contributing to the issue. please help to assign the issue.

jimy-byerley, 2025-06-08:
Sure why not, submit a PR when you are ready !

jimy-byerley, 2025-06-09:
The new packaging must also address #71 #100 , they are already addressed in the library code but in the current `setup.py`

jimy-byerley, 2025-07-05:
Hello, any news on this ?

PR #138 (2025-07-19): switch packaging to poetry
fixed #134 
this also allows us to ship linux prebuilt packages